bambulab/BambuStudio #670: [Feature Request] Bring back Lidar Flow Calibration Warnings

jaanshgo, 2022-11-19:
The printer used to report when the flow calibration didn't work. That was before the textured PEI plate was offered by BL and near when it was announced, all the flow warnings stopped appearing. 

We had calibration feedback and now we don't. Why remove such an useful feature? And silently at that. Most people that buy the official textured plate might run the calibration and see no warnings. Because of this, they might be wasting filament and time as the calibration could be failing every time and the printer using the default values.

The lidar calibration is touted as one of the top features this printer has and people have no idea if it works or not because there is no feedback. I want to know if the calibration worked in my textured plate so I know if I should skip it in that plate and do it only in the ones that produce no warnings. 

Bambu Lab, Please don't go out of your way to remove features and make the product worse. I understand if people buy the textured plate and get a warning every time they print, they might freak out or think there is something wrong with the printer but hiding if it's working or not is not the right way to go about things.

I love your printers. I have 3 and plan to buy more. I want BL to succeed so please do the right thing and don't move backwards even if it might receive some backlash from some customers that didn't read a warning before buying the textured plate.

hifihedgehog, 2022-11-20:
One suggestion might be to have a “verbose warning” option so minor warnings like these that Bambu Lab deems are unnecessary for their average end user can be restored for advanced users who want to see them.

psiberfunk, 2022-11-22:
Flow calibration is unfortunately pretty hit or miss and is not reliable at all.  Until bambu allows us to tune the accuracy of the flow calibration (by doing more lines with finer increments or larger values than the hard-coded ones), save values (do it once per filament, for example and then store it with a filament profile) and similar things.. it's mostly a gimmick that does not work reliably except maybe for their filaments..