bambulab/BambuStudio #671: [Feature Request] Time-based timelapse recording

kiliankoe, 2022-11-19:
I love timelapse shots of my prints, I pretty much always record them. But I also just love watching the printer do its thing filling layers and wonder if it would be feasible to add another timelapse mode that doesn't take a picture on every layer, but perhaps every second. Especially for very detailed flat prints this could be super cool to watch, where the current timelapse modes don't look all that interesting.

Thanks for considering! Keep up the awesome work! ❤️

iandouglas, 2022-11-21:
I would love to have an "add 5 seconds to the end of the video" feature as well

kiliankoe, 2022-11-22:
@iandouglas That's a great idea as well! If you're looking for a quick workaround for now, ffmpeg is great for stuff like this. The following command does exactly that, it just extends the last frame of the video for however long you wish, 5 seconds in this example.

```bash
ffmpeg -i path/to/your/video.mp4 -vf tpad=stop_mode=clone:stop_duration=5 longer.mp4
```