bambulab/BambuStudio #673: Setting colour in slicer with single colour print

rkarlsba, 2022-11-19:
Hi all

If I set the colour to number 2 in the slicer and press print, I still need to set it again in the filament selector when the print popup comes. Is it possible to remove this last step if I've set the colour in the slicer in the first place?

roy

MadMax389, 2022-11-20:
I believe if you match the color in the slicer to the color selected in the AMS device screen (regardless of the material selected) it will select the right AMS slot when sending the print. I will set up the colors is the AMS device screen first, then sync the AMS colors in the slicer. Then you can select a different material in the slicer and still maintain the proper color. When you send the print it selects the right AMS slot.
[image]
[image]

rkarlsba, 2022-11-26:
> print (Issue #673)

The problem is that this doesn't work with less than four colours, as far as I can see. I just sent a two-colour job to the printer and chose the correct colours. The print screen shows up and it automatically selects filament 1 and 2, regardless of what I chose in the prepare screen. This is the problem. 

Vennlig hilsen 

roy 
-- 
Roy Sigurd Karlsbakk 
(+47) 98013356 
-- 
I all pedagogikk er det essensielt at pensum presenteres intelligibelt. Det er et elementært imperativ for alle pedagoger å unngå eksessiv anvendelse av idiomer med xenotyp etymologi. I de fleste tilfeller eksisterer adekvate og relevante synonymer på norsk.

MadMax389, 2022-11-26:
Yes I see your problem now. I loaded two spools in slots 3 and 4 and assigned colors on the device screen (slots 1 and 2 empty). When you sync the AMS on the prepare screen it assigns the colors to slots 1 and 2 (instead of 3 and 4). Seems like a bug.

rkarlsba, 2022-11-26:
Correct. Hope you'll sort it out :) 

Vennlig hilsen 

roy 
-- 
Roy Sigurd Karlsbakk 
(+47) 98013356 
-- 
I all pedagogikk er det essensielt at pensum presenteres intelligibelt. Det er et elementært imperativ for alle pedagoger å unngå eksessiv anvendelse av idiomer med xenotyp etymologi. I de fleste tilfeller eksisterer adekvate og relevante synonymer på norsk.

MadMax389, 2022-11-26:
LOL. Not me, I'm just another user ;)