bambulab/BambuStudio #674: add Z-Offset setting

Woki-Leo, 2022-11-19:
My printer consistently makes the first layer too thin.
I have no way of adjusting that. Neither in the printer firmware nor in the Gcode.
Pls fix that.
[image]

Benderrodrigu3z, 2022-11-20:
The Current Fix is to run a Self-Test from the last tab on the left either that sets the Z offset or another thing is Run the Calibration from the 2nd or 3rd tab

I believe that the first one (Self-Test) sets the Z offset though

Benderrodrigu3z, 2022-12-21:
> We need this setting in order to offset PETG first layer. PETG sticks too well and being able to set a Z offset will resolve the issue. Even if it does not damage the bed it is a giant pain to remove the prime/calibration lines without raising the Z offset.

They said they added this in the new firmware, you have to add it to the start gcodeSincerely,Shane FSent from my iPhone13

suit, 2023-02-01:
you can set the z offset in the startup gcode - it is there for the P1P for example for the PEI sheet

closed 2023-02-06 by QingZhangBambu