bambulab/BambuStudio #675: A fatal error occurred: "Access Violation" when attempting to slice model

MadMax389, 2022-11-19:
Bambu Studio (v1.03.00.25) had been working fine for several days after I received my printer. Today, I started getting the error shown in the image when attempting to slice a model. Thereafter, no model would slice, even cube and cylinder primitives loaded on the plate. The program did not crash, I simply get the error when attempting to slice any model. I think I narrowed the problem down to user profiles I had created for filament and/or the printer. Troubleshooting steps:

1) Restarted Bambu Studio. Didn't work.
2) Rebooted computer. Didn't work.
3) Disabled antivirus. Didn't work.
4) Reinstalled Bambu Studio over the existing installation. Didn't work.
5) Uninstalled BS completely except for user profiles. BS automatically logged in to my account. Didn't work.
**6) Uninstalled completely including user profiles. Reinstalled it and did not log in. Model sliced normally.**
7) Logged in to my account in BS and selected the option to sync profiles with the cloud. Received access violation error.
**8) Had to uninstall and reinstall completely again. Logged into BS account but did NOT sync profiles. Model sliced normally.**
9) I recreated my profiles and the program is working normally.

I had created two profiles, one for filament and another for the printer. Could one of the profiles have gotten corrupted? It was working just fine for almost a week until it wasn't.

[image]

lanewei120, 2022-11-23:
hello @MadMax389 

thanks for reporting this issue~

1. if you select system profile or other user profile, is the slicing can be performed successfully?
2. could you help to upload your user profile which has the issue under folder
C:\Users\user.name\AppData\Roaming\BambuStudio\user\user.id\
we need to have a check for the values of that profile

for example the files:
[image]

MadMax389, 2022-11-23:
@lanewei120 
1. Sorry, I don't recall which profile was selected. Once the error occurred, no models (even primitives) would slice until the program was reinstalled. I don't recall if I tried to select a system profile after the error occurred.
2. I'm really sorry, but I had deleted the local profile and recreated it. It has been working fine ever since. I did disable "Autosyc Profiles" on the last reinstall, so the offending script might still be in the cloud account?

I can send you the account username via email if that helps.

lanewei120, 2022-11-23:
@MadMax389 
if enable the auto sync, the user settings will be downloaded to local directory again

and if this issue happens, no need to reinstall,
just need to select another preset(system preset or other user presets) from following menu
[image]

if you can reproduce this issue, please have a try and send the user preset to us
after that, you can also delete the preset which will cause this issue and create a new one

of course, you can send the username to us
however, that needs lots of time to check which preset will cause this issue

thanks

MadMax389, 2022-11-23:
@lanewei120 I will try this when I get a chance. It may be a couple of days because of US holiday.

MadMax389, 2022-11-26:
@lanewei120, I'm sorry but I was not able to recreate the problem. I did enable profile sync, but I had used the same name for the new profile as the old profile, so it must have overwritten the old one with the new one. If it happens again I will make sure to save the corrupted profile.

lanewei120, 2022-11-28:
@MadMax389 
ok, thanks very much for doing this

SwagDad99, 2023-01-16:
My issue seems to be the same, but only after I color the model.
Has this one been resolved, or just abandoned?

lanewei120, 2023-01-16:
> My issue seems to be the same, but only after I color the model. Has this one been resolved, or just abandoned?

@SwagDad99 
not solved, just not reproduced.

could you help to give more information as:
1. the detailed steps
2. the example 3mf
3. screen shots before and after slicing
4. the logs under C:\Users\user.name\AppData\Roaming\BambuStudio\log

lanewei120, 2023-06-29:
close it
if encounter this issue again, let's reopen it

closed 2023-06-29 by lanewei120