bambulab/BambuStudio #678: [Issue] Bambu Studio not starting reliable on dual Monitor setup (4k + 1080) under certain situation

BambuLabDiscord, 2022-11-20:
When bambu Studio is started on 4k Monitor,  placed on 1k Monitor full screen, and then closed on this screen. Bambu Studio will stuck loading on the  next startup attempt. Windows report with the typical  "Bambu Studio does not react" window.

To recover problem, the entire "\AppData\Roaming\BambuStudio" must be deleted.

FallenHorseman, 2022-11-20:
debug logs.
modifications done on this file  due privacy concerns:

-) deleted some hex values,
-)replaced User directory to UserName

[debug_Sun_Nov_20_13_43_41_29052.log.zip](https://github.com/bambulab/BambuStudio/files/10049707/debug_Sun_Nov_20_13_43_41_29052.log.zip)

bambu123, 2022-11-23:
There may be some app hangs reports and crash reports in Console. Please find and send that related to Bambu Studio to me.
The console is a system App you can search in the Applications folder.

FallenHorseman, 2022-11-23:
I assume you mean these files?
If this is correct, please be informed that in the english windows 10 version, it's called "Event Viewer".
If not can you send me a screenshot, to know what I've to look for?
[Event.zip](https://github.com/bambulab/BambuStudio/files/10079063/Event.zip)

bambu123, 2022-11-24:
Sorry, I thought it was macOS. Maybe we have to send you a debug build version to cache the hang stack. The minidump can be generated from the windows task manager when the program is hung, but it's 1

bambu123, 2022-11-25:
@FallenHorseman Hi, before using debug build to collect informations. Can you tell me something else?
Is it relate to the 4k monitor? if dis-connect this monitor, does it continue?
If remove the following line in the BambuStudio.conf , can Bambu Studio start normally?

--
        "window_mainframe": "399; 152; 1634; 937; 0",

FallenHorseman, 2022-11-30:
Sorry for delay.
Was absent from my PC for a while.
I played with the window_mainframe values.
not working: "window_mainframe": "-1928; 1072; 1936; 1096; 1",

working, but then it will be placed on my main Monitor (4k) again: "window_mainframe": "1928; 1072; 1936; 1096; 1", 

Here the requested dump:
https://drive.google.com/file/d/122j972K-e5Ms6PSnldS97slBmaVLBdLP/view?usp=sharing

Cannot unplug my 4k monitor as i then have no 2nd monitor.
As said, the application only freeze up if I place it on my 2nd screen and close it there.
If I take care that I am closing it on my 4k Monitor, it will startup fine next time on my 4k Monitor.

bambu123, 2022-12-01:
I found a injected dll (NahimicOSD.dll) caused this problem. It maybe from ASUS Sonic Studio. Could you update or uninstall it?

>	win32u.dll!NtUserEnumDisplayMonitors()	未知	非用户代码。已加载符号。
 	d3d9.dll!CEnum::GetAdapterMonitor()	未知	非用户代码。已加载符号。
 	NahimicOSD.dll!00007ffc0b8ae43b()	未知	非用户代码。无法查找或打开 PDB 文件。
 	NahimicOSD.dll!00007ffc0b8acc35()	未知	非用户代码。无法查找或打开 PDB 文件。
 	NahimicOSD.dll!00007ffc0b8aa232()	未知	非用户代码。无法查找或打开 PDB 文件。
 	gdi32full.dll!SwapBuffers()	未知	非用户代码。已加载符号。
 	BambuStudio.dll!00007ffbf571f1f1()	未知	非用户代码。无法查找或打开 PDB 文件。

bambu123, 2022-12-01:
Here is some more informations.

https://discourse.mcneel.com/t/rhino-freezes-on-secondary-monitor-problem-with-the-nihimic-audio-driver-on-the-motherboard/82258/37

FallenHorseman, 2022-12-05:
Thank you a lot. You are awesome!!
Uninstalling the mentioned tools resolved the issue

closed 2023-01-11 by SaltWei