bambulab/BambuStudio #680: Selecting/deselecting objects in slicer is extremely buggy

rkarlsba, 2022-11-20:
Attempting to select and deselect objects in the slicer where I have several objects available, works, somehow, but if I deselect one and just click on another to select or deselect that, it just selects the one I just deselected. It takes some time to manage to select/deselect objects and it seems random when this works or not. I'm on a mac with macOS 12.3.1 with the latest bambustudio, 01.03.00.25. Apparently, bambu studio also regularly hangs when I attempt to do this. Please fix  ;)

maggiesunh, 2022-12-07:
We can't reproduce it. Could you help to upload the failed 3mf files?

rkarlsba, 2022-12-07:
It happens to any model tested so far. When I click the tick box in the object list, it selects or deselects the object, but if I just move to another object and click it, it interprets this as a new click on the previous object, which reverses the last operation. This is one example:
[drittdusj.zip](https://github.com/bambulab/BambuStudio/files/10176975/drittdusj.zip)

SaltWei, 2023-03-02:
Does it still exist now?

SaltWei, 2023-03-02:
The v1.5.0 public beta has been released. Please ket us known if still exist. Thanks

closed 2023-03-02 by SaltWei