bambulab/BambuStudio #682: SUGGESTION: Disable keyboard shortcuts when add text tool is open

olithraz, 2022-11-20:
I can't tell you how many times I open the add text tool and end up with the text box not selected when I start typing. This causes keyboard shortcuts to be executed and usually ends up clobbering what I am working on. 

I think it would be a good idea to disable keyboard shortcuts entirely while the text tool is open to prevent this.

YifanWuBambu, 2022-11-23:
Hi @olithraz ,
Which shortcuts block your typing?

olithraz, 2022-11-23:
> Hi @olithraz , Which shortcuts block your typing?

Hi, sorry, I suppose I was not clear. 

My suggestion that when the add text tool is open, bambu studio would DISABLE all keyboard shortcuts. I find myself trying to type without having the text box selected but I have the add text tool open. Randomly typing into bambu studio would cause all sorts of things to happen. I feel like as a convenience thing, all keyboard shortcuts should be suppressed whenever the add text menu is open.

YifanWuBambu, 2022-11-23:
Okay. Got what you means.
Disabling all shortcuts may also make some other use confused. It is a hidden logic.
They may wondering in what senario the shortcuts do not work.

For this case, we may auto select the text box when you open text tool.

olithraz, 2022-11-23:
That would also be nice! I use it often to make sample swatches

--Brandon

YifanWuBambu, 2022-12-03:
@olithraz 
It will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu