bambulab/BambuStudio #683: AMS slots not

Benderrodrigu3z, 2022-11-20:
I have 12 AMS slots, if I leave slot 4 and slot 8 empty and then tell the BambuStudio slicer to sync instead of showing 12 slots it shows 10.... this is wrong on some many levels, is it going to pull from the correct line? is support filament still 12 or is it now 10 ?

[image]
[image]
[image]
[image]

Benderrodrigu3z, 2022-11-20:
The BambuStudio should still show slot 4 and say EMPTY with what color was last used in it since I told it to sync

Benderrodrigu3z, 2022-11-20:
and it self corrects on output? dang that's confusing
 
[image]

sjscymru, 2022-11-21:
I noticed the same issue when using a single AMS unit. I had slots 1,2 and 4 loaded but when I synced with Bambu Studio the filament in slot 4 was placed in slot 3.

My workaround was just to add the same flament again in slot 4 and use that for the print.

Agree with you that Bambu Studio should respect empty slots correctly.

YifanWuBambu, 2022-11-26:
Hi @Benderrodrigu3z @sjscymru 

There may be some concept misunderstanding.
"Filaments" - Means the filament used in a model project.
"AMS Slots" - Filaments that are installed in AMS' slots

You may use more filaments (such as 9 colors) than the slot count of AMS (such as 4 slots)  in a model project.
Plate 1 : used filament 1~4
Plate 2: used filament 5~7
Plate 3. used filamen 7~9
Each plate's used "Filaments" can be mapped to proper AMS slot when sending a print job to the printer.

Even you do not has an AMS, you can still create a multi-color project by scatter different color objects to different plates.
So, "Filaments" on the Studio's left sidebar is not the same as printer's AMS slots. And they are not one-to-one associated.
The AMS slot sync button is just designed for convinience. Users can quickly pick the AMS's filament settings(especially color and filament material) from AMS, instead of input them one by one on the Studio's sidebar.

Benderrodrigu3z, 2022-11-27:
that’s not the issue.The issue is that it doesn’t work with profiles that have support filament in say slot 12 with slot 4 missing, it breaks the profile and makes you go change the support slot every time you have any slot open after a pull, do I have to install it into slot 1 because yall won’t fix this?Sincerely,Shane FSent from my iPhone13

Benderrodrigu3z, 2022-12-03:
Yes there is some concept misunderstanding, how about you make it respect the fact that I spent over $1,000 on the AMS units

YifanWuBambu, 2022-12-03:
> it breaks the profile and makes you go change the support slot every time you have any slot open after a pull

Hi @Benderrodrigu3z , sorry for the late response. Let's make the issue more clear.

1. Could you explain more about "it breaks the profile"? 

2. By "you go change the support slot", you mean change the support index on the sidebar or on the "send to print" dialog?
[image]
[image]

3. Why do you have to change the solt manually everytime you have a slot open? In such case, the filament in slicer's side bar will be auto-mapped to the slot when sending a print job. You do not need to do manual assignment.
For example, my slot 3 is empty
[image]

After sync AMS slots to slicer's sidebar filaments, empty slot is ignored.
[image]

Set "support filament" to filament index 3 (Bambu Support W), slice and send to print. Slicer detected that the Bambu Support W is in AMS slot 4, so it will use AMS slot 4 to to print the support.
[image]

Benderrodrigu3z, 2022-12-03:
I just got word on case 572, looks like a similar issue will be fixed, my concern with this was say 3 or even 4 are missing then the numbers are way off to which slot

YifanWuBambu, 2022-12-05:
> my concern with this was say 3 or even 4 are missing then the numbers are way off to which slot making it more confusing to us the end use

Hi @Benderrodrigu3z , if we sync the empty slots to slicer's filament list (on left sidebar), such as slot 4, slicer cannot assiciate a valid filament preset (parameters) with it and the slicing procedure will encounter errors.

randyre, 2022-12-06:
@YifanWuBambu 
If there's no filament loaded, then the slicer should just ignore the slot unless it's bound to the model in the profile, then there should be an error saying "missing filament in slot X, the profile needs updating" or something like that.