bambulab/BambuStudio #684: Network access on Linux build

fleutot, 2022-11-21:
I cannot find my printer from my Linux machine.

BambuStudio built natively as per this [issue thread](https://github.com/bambulab/BambuStudio/issues/12).

I believe [this change set](https://github.com/bambulab/BambuStudio/pull/122) is merged, I am building on current master 96707fc.

The build works, I have the printer (X1 Carbon), my phone, and my laptop on the same SSID. The printer shows up on my phone, but not on my laptop (Ubuntu 20.04). I am logged in on the same account on all three devices.

The "Can't Find My Device" [page](https://wiki.bambulab.com/en/software/bambu-studio/failed-to-connect-printer) is not helpful. It says to scan the QR code, but I cannot do that on my laptop. I scanned with my phone, hoping it would show up on the laptop since logged in on the same account, but nope.

fleutot, 2022-11-21:
Well, actually no, I build on run the application successfully, but without building the dependencies. When I try to build them according to [this reply](https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1261724783), I get a whole load of patch failures:

```[ 90%] Performing patch step for 'dep_wxWidgets'
/home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:28: space before tab in indent.
    	, const wxWindow *window = nullptr
/home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:30: space before tab in indent.
    	);
/home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:41: space before tab in indent.
    	, const wxWindow *window = nullptr
/home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:43: space before tab in indent.
    	)
/home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:167: trailing whitespace.
        (anchorSideVertical & wxUP) || 
Checking patch build/cmake/init.cmake...
error: while searching for:
        else()
            find_package(OpenGL)
            if(WXGTK3 AND OpenGL_EGL_FOUND AND wxUSE_GLCANVAS_EGL)
                set(OPENGL_LIBRARIES OpenGL::OpenGL OpenGL::EGL)
                find_package(WAYLANDEGL)
                if(WAYLANDEGL_FOUND AND wxHAVE_GDK_WAYLAND)
                    list(APPEND OPENGL_LIBRARIES ${WAYLANDEGL_LIBRARIES})

error: patch failed: build/cmake/init.cmake:413
error: build/cmake/init.cmake: patch does not apply
```

deftdawg, 2022-11-22:
Don't build it natively, use podman to build it using the `ContainerFile` to create an AppImage and run that.

fleutot, 2022-11-22:
@deftdawg Unfortunately, podman is not available on Ubuntu 20.04.

deftdawg, 2022-11-22:
> @deftdawg Unfortunately, podman is not available on Ubuntu 20.04.

In that case, you can manually do every step of the Containerfile... The Containerfile is based off of 20.04 to use an older version of glibc for compatibility with Arch/Steam Deck and other old glibc systems. 

You'll need to build the deps per the Containerfile steps so that the proper newish version of wxwidgets gets built and patched.

lanewei120, 2022-11-23:
> Well, actually no, I build on run the application successfully, but without building the dependencies. When I try to build them according to [this reply](https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1261724783), I get a whole load of patch failures:
> 
> ```
> /home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:28: space before tab in indent.
>     	, const wxWindow *window = nullptr
> /home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:30: space before tab in indent.
>     	);
> /home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:41: space before tab in indent.
>     	, const wxWindow *window = nullptr
> /home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:43: space before tab in indent.
>     	)
> /home/gauthier/src/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:167: trailing whitespace.
>         (anchorSideVertical & wxUP) || 
> Checking patch build/cmake/init.cmake...
> error: while searching for:
>         else()
>             find_package(OpenGL)
>             if(WXGTK3 AND OpenGL_EGL_FOUND AND wxUSE_GLCANVAS_EGL)
>                 set(OPENGL_LIBRARIES OpenGL::OpenGL OpenGL::EGL)
>                 find_package(WAYLANDEGL)
>                 if(WAYLANDEGL_FOUND AND wxHAVE_GDK_WAYLAND)
>                     list(APPEND OPENGL_LIBRARIES ${WAYLANDEGL_LIBRARIES})
> 
> error: patch failed: build/cmake/init.cmake:413
> error: build/cmake/init.cmake: patch does not apply
> ```

this is just because the rebuilding of wxWidgets will patch again without clean previous patched result,
you need to cleanup the wxWidgets directory by build again:
cd deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets
git reset --hard
#you can also delete dep_wxWidgets-prefix directly

fleutot, 2022-11-24:
@lanewei120 Do you mean that the patched were already applied? I don't think I did it manually, but maybe the build process does it? In that case, if that step is not needed anymore, then the answer I linked to is obsolete. Maybe edit it to show this is not needed anymore?

lanewei120, 2022-11-24:
> @lanewei120 Do you mean that the patched were already applied? I don't think I did it manually, but maybe the build process does it? In that case, if that step is not needed anymore, then the answer I linked to is obsolete. Maybe edit it to show this is not needed anymore?

yes, each time it builds, the patch will be applied
from your log, it seems it is not the first build
so the patch were applied before

we will refine this logic, and currently you need to clean it to avoid this error for following build

lanewei120, 2023-01-31:
already fixed

closed 2023-01-31 by lanewei120