bambulab/BambuStudio #685: Resizing only in one direction is not possible

Obicom, 2022-11-21:
I worked long time with the Prusa slicer and there I was able to resize e.g. support blocker cube only in one direction (left or right / top or down) with pressing and hold CTRL button. I miss this function on Bambulab Studio slicer or did I have overseen something?

YifanWuBambu, 2022-11-23:
@Obicom 
You can resize in one direction by simply dragging the "cubic grabber" in the middle of lines.
[image]

Obicom, 2022-11-23:
@YifanWuBambu maybe a missunderstanding. If I do as shown bei you with the green cubic grabber in the front and pull it to me the front and the back site grows. Same with left and right. What I want is thar if I pull the green cubic grabber in the front in my direction that only the front grows but not the back site. In Prusa slicer it is possible with holding the CTRL key and select on of the cubic grabbers. The second one it then hidden (in the back). Please try it out.

YifanWuBambu, 2022-11-23:
Thanks for the explanation. We will add it in recent releases.

SaltWei, 2023-03-01:
Available in v1.5.0 Public beta

closed 2023-03-01 by SaltWei

chiwon575, 2024-01-07:
good afternoon i am still facing this issue.  is there any way to resolve it.  every time i click on any cubic grabber it stretches the model in both directions.   i have tried with ctrl and alt  shift etc with no success

bitfrickler, 2024-12-10:
Press CTRL while dragging.

vanowm, 2025-01-26:
> Press CTRL while dragging.

IMHO this should be the opposite, CTRL = resize from the center...

anyway, thanks, at least there is that now.

vanowm, 2025-01-26:
Actually, even with CTRL, when dragging the corner, it makes the opposite corner gray, like indicating it will be stationary, but it still resized from the center.