bambulab/BambuStudio #699: FTP is being used for data transfers

rkarlsba, 2022-11-23:
It seems FTP is used between BambuStudio and the Bambu printer, the one without encryption or anything, and being hard to handle with NAT, that is, impossible to handle with NAT if someone turns on encryption. This is a *major* security flaw and I don't know anyone in their mind that would use FTP in 2022. SFTP (the one that comes with SSH) has been around since 1995 and is very stable and secure. Also, HTTPS could be used without issues. FTP is a protocol that should not be used for anything, really.

This security issue probably originates from the Bambu firmware, but I couldn't find anywhere to report it, so I'll start here.

YifanWuBambu, 2022-11-26:
Hi @rkarlsba ,
Thanks a lot for your information. We are trying our best to improve the security.
For details, please refer to
https://blog.bambulab.com/answering-network-security-concerns/

SaltWei, 2023-10-08:
@sodachen22 Had this security issue been handled already?