bambulab/BambuStudio #700: Print Sliced File to All / Multiple Printers In one go

rhys100, 2022-11-23:
Had to print a file to 9 printers, takes a while in Studio to print 1 by 1, would be nice to have a bulk screen where I can print to all at once or select which printers on thye one screen and it prints to all.

That way I can load all the filament up on all printers prepare them and hit GO!!!

jkenneydaniel, 2023-06-14:
+1 on this - can we get an idea on priority here for this feature? Having more than 4-5 printers and trying to print to them all is extremely cumbersome, it'd be great to be able to select the printers I'd like to send a specific print to.

My thought on this:

1. Slice file
2. Select Print to x
3. Window is visually similar to existing Print to screen
4. Has a list of printers with a checkbox to select which printers to send print to
5. File is uploaded to cloud with list of printers and cloud distributes to selected printers

Currently, sending a print to 24 printers takes me almost 30 minutes of just clicking buttons - not a super efficient use of my time. This feature would be a _huge_ improvement!

lanewei120, 2024-06-26:
already supported

closed 2024-06-26 by lanewei120