bambulab/BambuStudio #702: Add printer in LAN mode by IP address

rkarlsba, 2022-11-23:
Since the WAN mode is [utterly insecure](https://blogg.karlsbakk.net/2022/11/23/bambu-lab-x1-carbon-the-flipside/), adding a printer by its IP address would be nice in LAN mode when mDNS isn't available. Either because the printer is on a different subnet and the router doesn't route multicast, which is quite normal or if you have an enterprise wifi system (for instance from Cisco), which also may drop mDNS.

c2h5oh, 2022-11-27:
Multicast being blocked between wired and wireless connections is fairly common even on consumer/SOHO devices, so this would be nice even just for better connectivity.

merasil, 2022-12-01:
+1
i have multiple vlans and adding the printer via ip would be awsome. mDNS is pain in the ass for subnets...

gashton, 2022-12-10:
+1
Not sure why this wasn't done in the first place... definitely a must-have!
Bambu appear to have implemented Multicast SSDP instead of mDNS, Which is a shame as Avahi does not support this.

rkarlsba, 2022-12-11:
From Wikipedia [SSDP] was formally described in an IETF Internet Draft by Microsoft and Hewlett-Packard in 1999. Although the IETF proposal has since expired (April, 2000),[1] SSDP was incorporated into the UPnP protocol stack, and a description of the final implementation is included in UPnP standards documents

Doesn't seem very sane to use with anything, tbh.

c2h5oh, 2022-12-11:
Even fallback to regular DNS would be an improvement - I can always setup a local DNS entry for the printer..

rkarlsba, 2022-12-11:
Which reminds me that IPv6 support would be nice. Seems the printer is running Linux after all, so it really can't be that hard.

merasil, 2022-12-11:
> +1
> Not sure why this wasn't done in the first place... definitely a must-have!
> Bambu appear to have implemented Multicast SSDP instead of mDNS, Which is a shame as Avahi does not support this.

That explains why i cant get the Printer working in lan mode in my vlan network… i should have used wireshark in the first Place…

rkarlsba, 2022-12-11:
> That explains why i cant get the Printer working in lan mode in my vlan network… i should have used wireshark in the first Place…

You can remedy that by enabling multicast routing or forwarding between VLANs if the router understands that sort of stuff. Still, adding by IP or DNS name shouldn't be very hard to fix and is the good solution to this.

R1N4x, 2022-12-15:
Lots of fixes and improvements with v1.4 but this one is still missing.

rkarlsba, 2022-12-16:
Can it really be that hard?

rehashedsalt, 2023-02-23:
With today's outage, this is now especially pertinent. There is _already an existing dialogue for this feature_ as of 1.4.2 -- there should be very few blockers for this now.

julie777, 2023-03-17:
My thoughts on this. 
It should not require an app using bluetooth to set up the printer for LAN mode. The selection for lan mode should always be available.
After selecting LAN mode only there needs to be a way to connect the printer to the network. A simple way would be to allow putting a file "network credentials.txt" on the SD card and the printer can use network name and password from that file to connect to the network.

When the printer connects to the network it should display its ip address and passcode for bambu studio to connect to it. I am not a network expert so I am not sure about broadcasting its availability on the network. I think this should be okay on a LAN, which makes it discoverable without having to type in the IP address in bambu studio.

kriordan37, 2023-07-19:
+1 i should be able to use static ip addresses and point bambu studio to that. this issue is preventing me from using the printer in lan mode with a vpn

tinuva, 2023-08-16:
With the latest issue happening now: https://www.reddit.com/r/BambuLab/comments/15rznpy/printer_started_printing_by_itself_damaged_itself/

This issue should really receive priority asap!

ellenhp, 2023-09-13:
I posted this in another issue but it looks like this is the primary issue for adding printers by IP address so I'm going to post it here too, sorry about the spam. I can delete the comment in the other issue if it's a problem :)

I got a little bit frustrated with this issue because having command and control in the slicer is a key feature IMO. So I reimplemented the networking plugin to add this feature (and others).

https://github.com/ellenhp/bambu-farm

timothyzalusky, 2023-10-03:
Another +1 for this feature.

fritzw, 2023-11-07:
> As a temporary solution. I've written a [script](https://github.com/gashton/bambustudio_tools/blob/master/bambudiscovery.sh) to generate an unsolicited response to the upnp/ssdp discover, to tell BambuStudio the IP of the printer. It's meant to be run on Linux, but may work on Windows too if run in a VM/Container/Cygwin.
> 
> It has to be run each time BambuStudio is restarted, as Bambu will forget until it receives the response to its M-SEARCH ssdp:discover.

I've [converted the SSDP script to PowerShell](https://gist.github.com/fritzw/7d981046b509a27e204d01c9bd73f37b), so that it can be used on Windows as well.

TonyAlberto, 2023-12-02:
> I've [converted the SSDP script to PowerShell](https://gist.github.com/fritzw/7d981046b509a27e204d01c9bd73f37b), so that it can be used on Windows as well.

I just tested it and it's ok, it works, thanks! I will only add that it may be necessary to change the port, in my case it is listening on port 1990 instead of 2021 (here is information about these ports: https://wiki.bambulab.com/en/general/printer-network-ports). The $TARGET_IP line itself is not needed in the powershell script because it is not used at all in this case.

hoegge, 2024-02-09:
It is quite crazy you cannot add a printer by IP - it means a lot of user simply cannot connect to it and means you can return the printer and get a refund, since it does not work.

ellenhp, 2024-02-09:
I imagine it's hard to find time to figure out the user experience, but if anyone from Bambu is reading this, I think I'd like to suggest that you add a way to configure printers by static IP in the config file, and worry about the slicer UI code for that configuration later when you have more time available. A lot of users would be okay editing a config file if it means not having to return the printer (or in my case, re-implement the entire networking plugin in Rust).

To anyone here who runs linux and likes fiddling around with semi-broken software, the [bambu-farm](https://github.com/ellenhp/bambu-farm) repo I linked earlier will get you up and running. It's not perfect. Immediately after a print it drops into an infinite loop on a worker thread so you have to close the slicer and restart the server-side program too, but it gets the job done for me and might be useful for someone else at their wit's end.

RagingRoosevelt, 2024-02-09:
Yeah, without this, the X1E that my employer bought doesn't work on our corporate network. I can ping the printer and it's on the same subnet as my computer, but because it's trying to auto-discover, I can't add it to Bambu Studio.

rehashedsalt, 2024-02-09:
>Yeah, without this, the X1E that my employer bought doesn't work on our corporate network.

Corporate environments are an especially pertinent example -- mDNS is oftentimes disabled as a basic security measure.

RagingRoosevelt, 2024-02-09:
> mDNS is oftentimes disabled as a basic security measure

That's exactly what's causing the issues I'm running into.  And for Bambu to sell the printer specifically on the merits of compatibility with a corporate environment, but not support this direct IP / hostname specification seems disingenuous at best.

walterwongbbl, 2024-02-09:
We are considering this feature, and recently it's the Chinese New Year holiday, so our responses might be slower. We apologize for any inconvenience caused.

rkarlsba, 2024-02-10:
Happy new Chinese year! But I wonder - it's been more than 14 months since I posted this. Does it really take that long to fix such a minor thing?

kluppy, 2024-04-04:
Without the ability to add printers via IP the software is not fit for purpose in a corporate environment.
I think I will be returning my X1E as such.

Kablahz, 2024-04-08:
+1. Currently have an X1E in a corporate environment and have had to resort to FTPS to allow users to upload 3mf files from Studio. Company wants to purchase additional X1E units but we have put this on hold until we can statically address these printers and not rely on mDNS(not going to happen) or FTPS. Please make this a priority, should have been implemented before the X1E release.

M1-Olsson, 2024-04-11:
So the X1E has the same problem, lol Enterprise edition with a extreme premium in price and still not working.

We could use another printer but I guess we stick to actual enterprise brands and not the hobby ones.

rkarlsba, 2024-04-11:
> +1. Currently have an X1E in a corporate environment and have had to resort to FTPS to allow users to upload 3mf files from Studio. Company wants to purchase additional X1E units but we have put [...]

As [gashton](https://github.com/gashton) wrote above, it's not even mDNS, it's SSDP, which isn't described in any valid IETF document, but later implemented in uPnP, but who on earth would allow uPnP on an enterprise network…

ColColonCleaner, 2024-04-24:
I hope this is fixed soon. Just got an X1C and cannot use it over my network because of this issue.

timothyzalusky, 2024-04-25:
Maybe we can light a fire under bambu lab by mentioning the competitors that have IP address printing...
Ultimaker
Makerbot
Creality
Formlabs
Markforged
I could go on.  We need this feature.

rkarlsba, 2024-04-25:
Octoprint
Fluidd
Mainsail
(anything, really, except Bambu)

ColColonCleaner, 2024-05-23:
It's really strange there is no movement or comment about this problem from Bambu. Not sure why they are seemingly ignoring it, the issue has been around for a long time now.

rkarlsba, 2024-05-23:
They are passively ignoring this, for reasons unknown. Perhaps they make their money out of cloud traffic?

M1-Olsson, 2024-05-24:
They have LAN mode so the could traffic is optional already. I use the script mentioned above and it works great so the problem is that the client can't find the printer itself unless on the same network. I can't understand why they just don't fix this, maybe we will have better luck with the Orca guys fixing this... 
I did see a windows once when trying to start the video stream asking for an IP but that seems only be for the video.

ColColonCleaner, 2024-05-24:
> They have LAN mode so the could traffic is optional already. I use the script mentioned above and it works great so the problem is that the client can't find the printer itself unless on the same [...]

The script doesn't work for me for some reason. Not sure why. Is there a trick to using it that I'm missing?
Printer never shows up.

The only way I can interact with the printer over the network is via FTP to drop gcode files in.

rkarlsba, 2024-05-24:
I don't think you can fix this with orcaslicer either - I can only guess it's the bambu connectors that are doing this - otherwise, it'd be replaced long time ago. These are closed source, and AFAICT, breaks AGPL quite clearly. I wrote a piece about that a year and a half ago and got an answer from Bambu that wasn't very comforting, but I didn't follow up on it. I'll see what I can do in the future.

ColColonCleaner, 2024-05-24:
@rkarlsba did you have to change that script at all to make it work, aside from filling in the printer attributes? It doesn't work for me.
I'll try running wireshark to see what port bambustudio is expecting and see if that's different, but otherwise I'm not sure why others are able to use it but it doesn't work for me.

ellenhp, 2024-05-24:
> I can only guess it's the bambu connectors that are doing this - otherwise, it'd be replaced long time ago. These are closed source, and AFAICT, breaks AGPL quite clearly.

This is correct, but I reimplemented the networking plugin in a weekend or so for [bambu-farm](https://github.com/ellenhp/bambu-farm). This is very fixable.

ellenhp, 2024-05-24:
Yeah I've dropped the link a few different times in case it's helpful to people, but like I mention in the readme it's a very rough piece of software, not at all something an average person can just pick up and use. Also only builds and runs on linux, but it might be a good starting point for someone who wants to fix this once and for all (probably in collaboration with orcaslicer)

timothyzalusky, 2024-06-13:
This is in the top 10 based on number of comments, but if you add up each separate issue thread.  It is well into the 100s of comments on this issue.  @walterwongbbl please give us a timeline (I was expecting this resolved in the first year, not two years later).

timothyzalusky, 2024-06-13:
Add printer in LAN mode by IP address #702
Printer IP Dialog #3714
LAN mode does not work between VLANs #3272
[Feature Request] Ability to add Printer by IP address (Lan Only Mode) #2426
Add device by IP address, without needing to trigger a connection error first #1549
Ability to enter IP of Printer(s) #2232
Fixed IP Bambulab X1 Carbon and P1P #1737

@walterwongbbl that is 7 separate issues (possibly more that I missed) that will be resolved by this.  Ulitmaker can do it so can you.

Alex-Schaefer, 2024-06-17:
One could think they want all people, including companies, to use their cloud and upload everything there. Why, and what they are doing with the files, is up to you to assume.

Difficult to believe, that the ability to enter an IP address is such a difficult task.

timothyzalusky, 2024-06-17:
They wouldn't have released the X1E that can be ran fully offline if that truly were their intent.  I would even pay more to have this capability.  Come on Bambu Lab!

Alex-Schaefer, 2024-06-18:
X1C and other printers can use the LAN only mode, too. And what I read, the SSDP discovery problem is the same with X1E: If the printer can not be detected by BambuStudio/OrcaSlicer, you can't use it.

M1-Olsson, 2024-06-18:
> X1C and other printers can use the LAN only mode, too. And what I read, the SSDP discovery problem is the same with X1E: If the printer can not be detected by BambuStudio/OrcaSlicer, you can't use it.

Correct, we use the scripts mentioned above as a workaround but that only makes this more comical as that solves the problem so the only thing missing is a way to enter the IP in the slicer and it will work....

rkarlsba, 2024-06-18:
My printer is down atm, waiting for parts, but I just wonder - has anyone tried to use https://github.com/initstring/evil-ssdp to intercept and perhaps spoof the SSDP?

elexx, 2024-06-18:
> My printer is down atm, waiting for parts, but I just wonder - has anyone tried to use https://github.com/initstring/evil-ssdp to intercept and perhaps spoof the SSDP?

Yes, see above :-)
* Bash script: https://github.com/bambulab/BambuStudio/issues/702#issuecomment-1347672299
* Powershell script: https://github.com/bambulab/BambuStudio/issues/702#issuecomment-1798553304

Alex-Schaefer, 2024-06-18:
> My printer is down atm, waiting for parts, but I just wonder - has anyone tried to use https://github.com/initstring/evil-ssdp to intercept and perhaps spoof the SSDP?

I created a python script derived from the mentioned bash script, you can use it to send a fake SSDP packet with the printer IP from your PC to your PC 😬 
https://gist.github.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3
It's linked in another similar issue, too: https://github.com/bambulab/BambuStudio/issues/2426#issuecomment-1700541962

rkarlsba, 2024-06-18:
> As a temporary solution. I've written a [script](https://github.com/gashton/bambustudio_tools/blob/master/bambudiscovery.sh) to generate an unsolicited response to the upnp/ssdp discover, to tell [...]

I made a few, tiny changes in a couple of PRs ;)

rkarlsba, 2024-06-18:
Btw, It should be possible to use the same method in OrcaSlicer to just spoof the SSDP if needed. Has anyone looked into that?

rkarlsba, 2024-06-18:
I thought about integrating something in orcaslicer to allow this without an external script

Alex-Schaefer, 2024-06-18:
I thought about implementing this in OrcaSlicer myself, too. But I'm lacking the time for it and am not very familiar with C++.
And honestly I don't think this should be something the customer should do, this should be the responsibility of BambuLab.

rkarlsba, 2024-06-18:
I cannot but agree, but if BambuLab doesn't care, which seems to be the case, it's better to have a good workaround than nothing.

ellenhp, 2024-06-18:
> I thought about integrating something in orcaslicer to allow this without an external script

If you're ok modifying OrcaSlicer you don't need to do any of the SSDP spoofing stuff. All the networking plugin does is (through a couple of intermediate steps) call `on_machine_alive` when it receives the ssdp packet. Call that function directly with [appropriately formatted JSON](https://github.com/ellenhp/bambu-farm/blob/main/bambu-farm-client/src/api.rs#L168-L183) and you're good to go.

rkarlsba, 2024-06-20:
Probably a five minute job if you know the code - I don't :P

theGidy, 2024-06-24:
Thanks for the workarounds, i will trying them out. Still small hope for bambu to fix this issue.  First bambu is also the last bambu if this will not be fixed.

tdgunes, 2024-07-19:
> We are considering this feature, and recently it's the Chinese New Year holiday, so our responses might be slower. We apologize for any inconvenience caused.

Hi there, are there any updates regarding this? @walterwongbbl @MackBambu

ookurogullari, 2024-07-19:
Looking forward to having LAN Mode on my P1S! Is there any expected delivery date for it @walterwongbbl @MackBambu ?

brbubba, 2024-07-23:
I see there's a flag for whether the printer is bound to a cloud account. I had to bind to an account for firmware update. Will the flag cause a refusal to connect or any other negative results? How do you unbind?

[47 more replies]