bambulab/BambuStudio #703: estimated time way off

PowerChaos, 2022-11-23:
hi,
this is just a small bug report
it has some influence to the workflow but it is not really critical

i got the following print https://www.printables.com/model/13216-trick-gift-box/files
[image]

and instead 2 hours it is already printing for 6 hours
so the estimated time is off by 3 times as much

i also founded the problem why it is off that much ( when i was creating this report ) 
in the slicer i got it selected as "Generic PA" by mistake.
but it got "generic PLA" loaded so it use the default pla settings ( nice failsafe ) ( 200 degree / 45 heatbed ) 

would be nice if it gave a warning , or that the time get updated to reflect the new change.
still i am happy it got this kind of failsafe , pla on PA temps would ruin a lot xD

Thank you
Greetings from PowerChaos

SaltWei, 2022-12-01:
Could you share this 3MF(absolutely same with yours) to us to reproduce this issue?

PowerChaos, 2022-12-01:
@SaltWei 
> Could you share this 3MF(absolutely same with yours) to us to reproduce this issue?

in meantime the version of the slicer is updated , so the time is a bit longer now 
Be sure to set the AMS to 4 times PLA 

This mistake of mine happend when i was not using the AMS before when i printed nylon ( generic PA )  and i forgot to put it back to PLA, when only PLA was loaded in the AMS  ( PA was set to slot 1 ), as my other prints where using slot 2 ( other color ) 

[image]
[main_body-pieces.zip](https://github.com/bambulab/BambuStudio/files/10129143/main_body-pieces.zip)

Thank you
Greetings from PowerChaos

SaltWei, 2022-12-02:
Since this is single color model, so I printed directly and checked the real printing time.
And the real time is close to estimated time.
Could you give more information about your printer firmware version?

PowerChaos, 2022-12-03:
@SaltWei 
the printer version is now the latest version ( 01.01.01.00 )
i think this was on a older version , but never got a update , so finally after a reboot it showed the update so i updated.
i think i also figure out what exactly went wrong as i can not reproduce it anymore.

it could be one of those 2 mistakes

1. the slicer was letting me choose a slot or let me print when it contained the wrong profile
2. The slicer used default profile after a update , and i dit not noticed it that it was on "default PLA" instead of "Generic PLA"

i just checked again with the slicer , and if i open 2 slicers it now seems that the "default" is set to Bambu PLA instead.
i was also using a older version of the slicer ( i use now 01.03.01.01 for slicer version, think it was 01.03.00.25 or 1 version older ) 
 ) 
 
 as there is no way "yet" to see the older files , else i could look at the sliced file gcode to see what version/setting where used
 
 Thank you
 Greetings from PowerChaos

Lesani, 2023-01-03:
I have the same issue, the printed time told by the slicer vs. the actual time is VASTLY different, on a multimaterial print though.
Time given is 42h (1 day 18 hrs)
Time given by the printer after the print (onthe display) is also 42h

I started the print on January 1st 13:34
It lost ~5min when the poop chute clogged somewhere in the middle, otherwise no stops (ps: thanks for informing about issues in the app!)
The print finished on January 3rd 19:59
Actual printing time was 54,5h (2 days 6 hrs 25 min)

that is a 30% difference! Unacceptable when you fully control both the machine firmware (and all its acceleration & speed values), the material changer and the time it takes to do the cutting, switching, flushing, as well as the slicer.

[image]

[Rocinante_BambuX1C_5Plate_PLA_plate_4.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/10339572/Rocinante_BambuX1C_5Plate_PLA_plate_4.gcode.3mf.zip)

Lesani, 2023-01-05:
[image]

[Rocinante_BambuX1C_5Plate_PLA_plate_3.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/10350181/Rocinante_BambuX1C_5Plate_PLA_plate_3.gcode.3mf.zip)
calculated to be 15h44m
reality turned out 19h 20m

Lesani, 2023-01-05:
@YifanWuBambu @SaltWei Looking at a sample size of 3 prints where I know the print itself went without issues (no pauses because of errors or such) if I divide each time difference by the amount of filament changes I arrive at 30seconds. could it be that you forgot to add the time it takes to change a filament into the calculation?

SaltWei, 2023-01-05:
This is normal for multi color with AMS.
Because sometimes AMS may fail to switch filament and it will retry.
Time estimation can not handle this situation when slicing and always consider the filament switching to be successful once.

Lesani, 2023-01-05:
@SaltWei  you are incorrect
I have not observed a single failed loading since installing a curve relief (https://www.printables.com/model/338542-x1-ptfe-tube-guide-to-fix-loading-issues-best-of-b) on the print head
and that on 3 different prints the time difference turned out to be exactly 30 seconds, already should tell you that this is not a "sometimes something happens and that increases time" issue, but a regular time difference.

SaltWei, 2023-01-05:
> Material 4 in AMS2 with a slightly longer PTFE tube (2x original)
The length of tube also influence the time of T command, and this information can't be got when slicing and estimating in studio, and this can be better tuned by user here.
[image]

Lesani, 2023-01-05:
correct, the longer ptfe accounts for ~1-1.5sec of additional time on that 2nd AMS, not 30sec though

if I modify to 32s load and 31s unload time, the calculations are correct. That tells us that you are missing 30sec somewhere else in the loading process, or the loading actually does take 30 seconds? What is defined as "filament load" and "unload" time? the time from AMS movement until it reaches the head? or the time the lead moved last, until it moves again?

regardless, you are missing 30seconds somewhere in there per filament change, and I will verify on subsequent (error-free) AMS prints that it is actually 30 seconds

Lesani, 2023-01-05:
Additionally, saying this needs to be tuned by the user on an extremely controlled machine as yours is outrageous.

closed 2023-06-13 by SaltWei