bambulab/BambuStudio #704: [bug] [macOS] Multiple instances of the app

tokamac, 2022-11-23:
On a Windows PC, multiple instances of the same program can be opened at once as separate processes. On the contrary on a Mac, Apple Human Interface Guidelines prohibits such a behavior to prevent cluttering of the user interface. In macOS, an app should always have a single icon in the dock, but multiple concurrent windows on screen inside that app if applicable, in the same unique process.

This has been the case for all Mac applications for decades. Alas lately, cross-platform development (led by Java, Electron, QT…) has introduced some perverse setback in the Mac world by ignoring some basic UI guidelines of macOS. This includes for example non-native Java menus, or even a misplaced menu bar (e.g. in Ultimaker Cura 5.x). 

In BambuStudio, the issue is that opening several 3MF files one after the other opens multiple concurrent processes of the app at once, multiplying the BambuStudio app icons in the Dock in series, instead of opening several windows in the same app process as it should be. Resulting in the Dock being unnecessarily cluttered:

[image]

This is not a critical bug, but there is room for improvement especially on the sides of usability and consistency of the user interface.

bambu123, 2023-01-11:
Seems not easy to fix, we are really open to multiple processes instead of multiple windows.

aharpole, 2026-09-15:
Hi there! Long time user of Bambu Studio on macOS.

Bambu Studio's current behavior for handling multiple documents in macOS is really not platform appropriate and I really hope that you can update Bambu Studio to work like a normal Mac app where there is one instance of the app running in the Dock, and under that app instance there can be multiple windows with multiple documents.

I finally got annoyed enough that I took the liberty of throwing together a proof of concept, rebased against the current master here: https://github.com/aharpole/BambuStudio/tree/feature/macos-multi-process

It is working and I am using it on my own machine, but it does make fairly sweeping changes to how Bambu Studio works (the biggest part being that it eliminates the assumption that there is one project open for the process). I would not recommend merging it as-is, but rather it's worth treating as the start of a conversation about how to properly approach this.

I tried to explore an alternative approach where each window maintained its own process so that the code could stay similar to the way it is now but we get the correct macOS behavior but I was not successful; it's not clear to me that macOS has a way for an AppKit window to own its own separate process.