bambulab/BambuStudio #705: [Feature Request] Turn off fans on pause

TheCableGuy99, 2022-11-23:
**Is your feature request related to a problem? Please describe.**
I'm not sure if this happens when you manually pause, but when the printer pauses for a filament runout the fans seem to keep going on full speed constantly. This can be even if you are away for hours. Considering the hotend cools down there's no point to it. In all cases where a print is paused the fans should be stopped automatically after a couple of minutes.

**Describe the solution you'd like**
Any time the printer is paused and hotend cooled down pause the fans when the hotend reaches a certain temp and the fans are not needed any more.

**Describe alternatives you've considered**
I can't think of any, it's pretty simple.

**Additional context**
It's all there, nothing to add.

Benderrodrigu3z, 2022-11-23:
kind of true, however a sudden increase and then decrease in temperature could detach the build from the plate
think about the bed, if its at 55c then the fan stops its going to go up to 60c before seeing it needs to lower the volts and then stabilize, since that power has to stay on the bed the motherboard in the back has to stay cooled from heat and capacitors desoldering from the board under the heat load, the hotted fan would be the only fan that you could turn off and even that is barely using electricity

TheCableGuy99, 2022-11-23:
I'm not saying turn the bed heater off completely, but the part cooling fans are needed for when printing, I don't think they need to be on at 100% constantly when not printing when I might not be back to the printer for hours because it's paused due to a filament issue (could be anything, not just running out). Just my pennies worth.