bambulab/BambuStudio #706: [Automatic bed detection at the lowest level] Bed Grinding when at lowest level and starting a print

BambuLabDiscord, 2022-11-23:
If you print out something large, and the print bed goes all the way to the bottom, then when you go to print something else the printer moves the print bed lower as a default. If the print bed is already at the lowest then it makes an awful grinding sound and does not detect that it is already at the lowest possible height it could be.

candre23, 2022-11-24:
I have noticed this issue as well, and it definitely needs fixing.  It's only a matter of time before skipping teeth wears out the belt or possibly damages the Z motion system.

sodachen22, 2022-11-26:
@BambuLabDiscord @candre23 
Thanks for the bug reporting.
1. The current has been set to a low level when automatic bed detection. According to our testing result, it will not break anything. 
2. We will fit it in the future

JPotter73, 2022-11-27:
Related: https://github.com/bambulab/BambuStudio/issues/647

ashleyuk90, 2022-12-27:
Is there any update on this? As much as I know auto bed level is set to a low current and should just cause the stepper to skip steps, it still makes a horrible noise and is causing unnecessarily wear on hardware.

sodachen22, 2022-12-28:
According to our testing result, it will not break anything. The solution for the optimization is still not finilized.

LuSEllDen, 2023-02-07:
@sodachen22 Solution can be to use the bed sensors. Move the bed up and if there will be increased preasure, just lower it down and then do the homing.

c2h5oh, 2023-02-08:
> @sodachen22 Solution can be to use the bed sensors. Move the bed up and if there will be increased preasure, just lower it down and then do the homing.

Simpler fix (entirely on printer side): 
- save bed height at the end of a print
- when new print is received check if last bed height is below a safe value - if it is raise it to safe height (by min_safe - current_height) before executing gcode of incoming print

LuSEllDen, 2023-02-08:
@c2h5oh That can be a solution, but that would mean the printer would write every last bed position when it's not printing. I am used to adjusting the height of the bed after printing and this can be done only in steps (to safely remove print, to clean the PEI...), so the position would have to be saved after each bed movement. Of course I don't go all the way down when I want to turn off the printer.

LuSEllDen, 2023-02-08:
> That can be a solution, but that would mean the printer would write every last bed position when it's not printing. I am used to adjusting the height of the bed after printing and this can be done [...]

candre23, 2023-02-08:
Here's the thing:  The printer is currently able to move the bed up, recognize resistance, and stop before skipping any teeth on the Z belt.  It does this repeatedly during homing and leveling.  Whatever speed and resistance sensitivity is used when moving the bed up during homing should also be used when moving the bed down slightly at the beginning of the procedure.  If the bed hits the bottom of the enclosure, it should stop trying to go down - just as it stops trying to go up when it hits the nozzle. 
 This is the simple and obvious solution.

ashleyuk90, 2023-02-09:
The printer uses the force sensors also built into the bed when doing the homing / bed levelling, these wouldn't have any benefit when the bed is at the bottom hitting the bottom of the printer.

However as they do with the layer skipping by monitoring the motor resistance they should be able to use this to detect the Z skipping fairly quickly and at least cut the noise/issue down to a second or so.