bambulab/BambuStudio #708: Anchor Solid Infill by X mm from SuperSlicer

djngnear, 2022-11-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I love the option to fill in small areas with solid infill on top surfaces like under text or smaller objects.  I go to Superslicer for this very reason. By default it's set to 150% but I like to bump it up to 300 or 400%. That way the monotonic infill just looks better and it doesn't need to navigate around holes.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Implement the option to change the solid infill percentage like in SuperSlicer.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

user897943, 2022-11-30:
Can you post two photos, showing without this option and with your option at your preferred setting.

djngnear, 2022-11-30:
Yeah so here is a sign with the default setting of 150% anchor solid infill (1st picture). 
And what it looks like at the top surface layer (2nd picture). 
By changing that setting to 300%, there's no more holes in the top surface to navigate around (3rd & 4th picture).
I think this setting is super useful for an extra nice top layer under lettering and smaller geometry.

[image]
[image]
[image]
[image]

user897943, 2022-11-30:
Thanks for this, it should definitely be in Bambu Studio. It would add strength in some situations I can think of. But not sure if they should name it like that... Just from reading I can't really make out what "anchor solid infill by" means, your example is great. What would you call the feature if you had to give it another name that's easier to understand?

AbeFM, 2022-12-21:
Outer (top) surface small hole solid infill ?

Yeah, I like what he's asking for!

dom2587, 2023-01-26:
im also looking for this setting to be enabled, i'm trying to print the head(amame) headset and there cones require the ability to set this to 0 for it to print correctly.

djngnear, 2023-01-26:
> im also looking for this setting to be enabled, i'm trying to print the head(amame) headset and there cones require the ability to set this to 0 for it to print correctly. 

Could you take a screenshot of what you're referring to? I'm interested.

dom2587, 2023-01-26:
> 

sure thing https://imgur.com/a/nZ7JCm8

if you look at the infill in bambu and compare it to the prusa slicer that is the infill anchor setting at play, why the bambu doesnt get the solid walls from the 3mf while prusa does is my next problem to sort lol.

djngnear, 2023-01-26:
Wow, yeah I see that. Hopefully they'll address it soon.

djngnear, 2023-08-14:
Please look into this!

SaltWei, 2023-08-15:
> im also looking for this setting to be enabled, i'm trying to print the head(amame) headset and there cones require the ability to set this to 0 for it to print correctly.

This can be set in latest bambustudio 
[image]

SaltWei, 2023-08-15:
> Implement the option to change the solid infill percentage like in SuperSlicer.

We will evaluate this feature and add it into developer mode. Thanks everyone!

djngnear, 2023-08-15:
> > im also looking for this setting to be enabled, i'm trying to print the head(amame) headset and there cones require the ability to set this to 0 for it to print correctly.
> 
> 
> 
> This can be set in latest bambustudio 
> 
> [image]
> 
> 
Yeah, that's for the sparse infill