bambulab/BambuStudio #714: Objects names not visible in object mode

Ruiaxe, 2022-11-26:
Bug:
When switching from global to object mode the objects in the plate have no name. This is a new issue with the latest patch.

To Reproduce_
Steps to reproduce the behavior:
1. Go to Objects mode

Expected behavior:
The object name should be available

Screenshots
[image]

Desktop (please complete the following information):
 - Windows 10
 - Version 22H2 Build 19045.2251

Ruiaxe, 2022-11-26:
Another small detail, the stl objects where inside a 3mf that was not made for Bambu Studio, only geometry was loaded.

YifanWuBambu, 2022-11-30:
@Ruiaxe 
I dragged a stl into Studio, but did not reproduce the issue.
[image]

What is the name of your stl? Could you share your stl with us?

Ruiaxe, 2022-11-30:
Hi! Just opened the 3mf in other slicer and the issue is the stl files have a very long name.

Attaching the 3mf and a screenshoot in prusaslicer.

[image]
[the-hardest-easy-game-l-game-model_files.zip](https://github.com/bambulab/BambuStudio/files/10124011/the-hardest-easy-game-l-game-model_files.zip)

YifanWuBambu, 2022-11-30:
@Ruiaxe 
Thanks for the 3mf. We will fix it.

YifanWuBambu, 2022-12-16:
This is included in v1.4.