bambulab/BambuStudio #715: Auto arrange puts large objects partially outside of the plate even if they fit in build area

c2h5oh, 2022-11-26:
**Describe the bug**
If you put a large object with size close to the entire X or Y of build plate auto arrange will put it mostly outside of the build plate. The attached file is an extreme example for the sake of being easily reproducible, but I believe I've seen it for objects with X or Y larger than 90% of print area.

**To Reproduce**
Steps to reproduce the behavior:
1. Add [Gridfinity Magnet Light Baseplate 3x6.zip](https://github.com/bambulab/BambuStudio/files/10097203/Gridfinity.Magnet.Light.Baseplate.3x6.zip) to build plate (it's 252mm wide)
2. Click on Arrange All Objects
3. Click Arrange
4. See error

**Expected behavior**
Automatic arranging should work correctly.

**Screenshots**
I can do one better - here's the video
https://user-images.githubusercontent.com/446070/204112703-cd977d79-e179-4ddb-b738-1486f690d387.mp4

**Desktop (please complete the following information):**
 - Windows
 - 1.3.1.1, but it was also present in at least one release before that (I haven't used earlier ones)

alexzogh, 2022-12-02:
Yes, very annoying bug. I've just reverted to manually placing everything

manch1n, 2022-12-02:
@c2h5oh cause spacing feature will inflate object before auto arrangement. So disable brim and lower spacing could solove this problem
[image]

c2h5oh, 2022-12-02:
@manch1n thanks for the workaround, but it doesn't change the fact it should be fixed as a bug.

tangketan, 2022-12-03:
@c2h5oh It's not a bug, but rather a feature request: when there is only one object, arranging spacing should automatically set to 0
I'll try to add it in.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu