bambulab/BambuStudio #717: Allow edit of print sequence by object

JPotter73, 2022-11-27:
It would be nice to change the order of the print sequence when printing by object.

For instance: If I'm printing 3 objects and I know that one is more risky than the others, I may like to put that one last. If it fails, I would still potentially get the two other prints.

JPotter73, 2022-11-27:
Case in point: there are 3 objects in this print and I was worried the 2nd may fail. I'm quite sure the 3rd would print.

Now I'll start a print for the last two objects, but maybe not since I don't have control over which will print 1st
[image]

YifanWuBambu, 2022-11-30:
It will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu