bambulab/BambuStudio #718: Auto arrange will incorrectly split print into multiple plates

c2h5oh, 2022-11-27:
**Describe the bug**
If several larger objects are added to the plate auto arrange function will split them into multiple plates, even if they easily fit on one plate, even leaving calibration areas clear

**To Reproduce**
Steps to reproduce the behavior:
1. Add all 3 to plate: [Downloads.zip](https://github.com/bambulab/BambuStudio/files/10098148/Downloads.zip)
2. Click on arrange all objects
3. Click on arrange - settings don't matter
4. See error

**Expected behavior**
Arrange uses minimum number of plates to arrange objects

**Screenshots**
Video of this happening: 
https://user-images.githubusercontent.com/446070/204132428-22bd0a84-1b28-4318-aa87-da2bb10aad82.mp4

Screenshot demonstrating those objects easily fit on plate:
[image]

**Desktop (please complete the following information):**
- Windows, Linux
- 1.3.1.1, but it was also present in at least one release before that (I haven't used earlier ones)

tangketan, 2022-12-03:
Our next release has solved this problem. The following is the arranging result:
[image]

closed 2022-12-03 by tangketan

c2h5oh, 2022-12-03:
@tangketan Thank you.