bambulab/BambuStudio #719: Please Support Rafts

davidjdixon, 2022-11-27:
**Is your feature request related to a problem? Please describe.**
When using more awkward filaments (eg Colorfabb Stonefill etc) that have relatively poor bed adhesion (even with gluestick), especially when printing a model that only has a tiny footprint. It's difficult to get very small 'dots/lines' to adhere.

**Describe the solution you'd like**
This is one of the primary reasons for a raft (beyond flat bed issues); a large, uninterrupted surface that is easy to put down, and then allows much more detailed items to adhere to it.

**Describe alternatives you've considered**
Brim's are useful, but they are primarily for keeping a print adhered during long prints on small footprints; they can work for some cases described above, but they also struggle when the radius of the raft itself is very small in the situation mentioned above, therefore they are not a consistent solution for very complex models using less standard filaments.

**Additional context**
Basic features such as rafts should be present in the software, especially since, at present, Bambu Studio is the ONLY slicer capable of interacting directly with the Bambu X1/P1P.

aridShrub, 2022-11-27:
I also have used raft for embedding hardware, for example a 3d printed handle with an embedded bolt, the threaded end goes through the raft. I agree with the use case of supports as well. On Pruslicer the raft would close down on the donut hole of the middle of the model by exactly one perimeter. But embedding with raft opens up a lot of possibilities, especially with support filament.

YifanWuBambu, 2022-11-30:
This will be included in the next release.

closed 2022-11-30 by davidjdixon

YifanWuBambu, 2022-12-16:
This is included in v1.4.