bambulab/BambuStudio #721: The printer lacks basic features found on Klipper

user897943, 2022-11-27:
**Describe the bug**
I originally posted on Reddit.

https://www.reddit.com/r/BambuLab/comments/z68tbo/x1carbon_firmware_wish_list/

I received my printer today and I am so far satisfied.

On Klipper and my Pi4 I could directly access the Mainsail webcam stream using the following URL format:

http://10.0.0.xxx/webcam/?action=snapshot

Where the IP is the local IP of my Pi4 running Klipper/Mainsail.

I could also use MQTT to control the printer.

I want this available on the X1C too, because I want to integrate it with Home Assistant for alerts etc. I can use the camera stream to train my own AI models too if I wanted to, using TensorFlow processing/Coral TPU via USB port.

I consider the printer unfinished until these features are provided. I think it's fantastic as a printer but please do not lock it down so much. Streaming and MQTT are already baked into the firmware, you just need to release a local endpoint for the LAN.

Edit to add: Also, one thing I miss from Klipper is the ability to cancel a specific object on a plate with multiple. Sometimes a small part will detach from the plate and I want to be able to tell the printer to ignore that part but print everything else, so that I don't have to throw away what has been printed so far because a small part caused an issue. Currently, the printer will just continue to extrude when there's nothing underneath it (because the object was detached from the plate during a print) and that will cause spaghetti.

Edit2: Maybe I jinxed myself but I had this exact issue just now! Was printing the plate stand from Printables, and the PTFE release tool, and 5 of those bambu cable+ptfe clips, one of the clips detached from the pei sheet, then it created spaghetti, I had to pause on each layer and remove manually to avoid losing the print. So that is not smart! PLEASE BAMBU LAB IMPLEMENT ASAP A WAY TO CANCEL ONE OR MORE OBJECT/S ON THE BED DURING PRINT WHEN THERE ARE MORE THAN 1 SEPARATE OBJECT ON THE PLATE!!!! I was about to lose an entire large object about 80g of plastic because of a tiny clip less than 2g... it's ridiculous, please please please put this in the firmware.

As for the slicer, where is the option to "multiply" an object? If I want to load one small part and set the number of instances to 30 to print 30 of those, there's apparently no easy way, I have to select the object and Ctrl+C then Ctrl+V 30 times!?!?!?! Come on... "fill bed with object" and "multiply by X" should be basic options please.

Another thing on the slicer, what's the point of having a text tool if it does not snap to another object or conform to curves? I wanted to make a ring with my friend's name on the inside as a negative of the text, but I could find no way to make the text conform to the curve on the ring. Please offer such option if possible, I know it's a little "exotic" but there's many examples I could give where text would need to conform to the shape of the surface I'm putting it on.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

YifanWuBambu, 2022-11-30:
@user897943 
"A̶s̶ ̶f̶o̶r̶ ̶t̶h̶e̶ ̶s̶l̶i̶c̶e̶r̶,̶ ̶w̶h̶e̶r̶e̶ ̶i̶s̶ ̶t̶h̶e̶ ̶o̶p̶t̶i̶o̶n̶ ̶t̶o̶ ̶"̶m̶u̶l̶t̶i̶p̶l̶y̶"̶ ̶a̶n̶ ̶o̶b̶j̶e̶c̶t̶?̶ ̶I̶f̶ ̶I̶ ̶w̶a̶n̶t̶ ̶t̶o̶ ̶l̶o̶a̶d̶ ̶o̶n̶e̶ ̶s̶m̶a̶l̶l̶ ̶p̶a̶r̶t̶ ̶a̶n̶d̶ ̶s̶e̶t̶ ̶t̶h̶e̶ ̶n̶u̶m̶b̶e̶r̶ ̶o̶f̶ ̶i̶n̶s̶t̶a̶n̶c̶e̶s̶ ̶t̶o̶ ̶3̶0̶ ̶t̶o̶ ̶p̶r̶i̶n̶t̶ ̶3̶0̶ ̶o̶f̶ ̶t̶h̶o̶s̶e̶,̶ ̶t̶h̶e̶r̶e̶'̶s̶ ̶a̶p̶p̶a̶r̶e̶n̶t̶l̶y̶ ̶n̶o̶ ̶e̶a̶s̶y̶ ̶w̶a̶y̶,̶ ̶I̶ ̶h̶a̶v̶e̶ ̶t̶o̶ ̶s̶e̶l̶e̶c̶t̶ ̶t̶h̶e̶ ̶o̶b̶j̶e̶c̶t̶ ̶a̶n̶d̶ ̶C̶t̶r̶l̶+̶C̶ ̶t̶h̶e̶n̶ ̶C̶t̶r̶l̶+̶V̶ ̶3̶0̶ ̶t̶i̶m̶e̶s̶!̶?̶!̶?̶!̶?̶!̶ ̶C̶o̶m̶e̶ ̶o̶n̶.̶.̶.̶ ̶"̶f̶i̶l̶l̶ ̶b̶e̶d̶ ̶w̶i̶t̶h̶ ̶o̶b̶j̶e̶c̶t̶"̶ ̶a̶n̶d̶ ̶"̶m̶u̶l̶t̶i̶p̶l̶y̶ ̶b̶y̶ ̶X̶"̶ ̶s̶h̶o̶u̶l̶d̶ ̶b̶e̶ ̶b̶a̶s̶i̶c̶ ̶o̶p̶t̶i̶o̶n̶s̶ ̶p̶l̶e̶a̶s̶e̶. "

**>> You can right click an object and choose "Clone" in the context menu. Then just enter the count that you want.
[image]
We have optimized the copy behavior, all copies shares the same mesh and they just slice once.**

"Another thing on the slicer, what's the point of having a text tool if it does not snap to another object or conform to curves? I wanted to make a ring with my friend's name on the inside as a negative of the text, but I could find no way to make the text conform to the curve on the ring. Please offer such option if possible, I know it's a little "exotic" but there's many examples I could give where text would need to conform to the shape of the surface I'm putting it on."
**>> This is in our plan, but may be not in recent releases.**

First thing - I am very disappointed that the 0.6 and 0.8 nozzles were neglected in terms of profiles. The 0.4 nozzle has 6 profiles ranging from extra fine to extra draft, and an Optimal profile that marries speed and quality together. The 0.6 nozzle has one profile and it's not even that great. The 0.8 also has one profile and it's also not great either. As a customer who bought the 0.6 and 0.8 I expected to see more quality, more profiles for each nozzle. Bambu Lab PLEASE show love to your 0.6 and 0.8 nozzles! I will post a picture to show calibration cube results and you can see that 0.6 and 0.8 using the only profiles available for them on Bambu Studio (01.03.01.01) are passable at best. Thankfully no print failures but the quality leaves a lot to be desired. Don't show love just to the industry standard 0.4, if you want your printer to be the best then ALL the accessories must work to the same quality level one expects after using 0.4mm nozzle (within constraints of course).
**>> We will check it and solve the issues in recent release. Thanks for reporting that.**

YifanWuBambu, 2022-12-01:
@user897943 
Could you share your 3mf with us?
[image]

alexzogh, 2022-12-02:
Besides the printing issues (which I do not have on my .6 or .8 nozzles), this should be relabed feature request. I would rather see the Bambu team focus on the MANY actual bugs before adding more features

SaltWei, 2022-12-02:
Please raise different issue on different issue ticket.For tracking issue easily

SaltWei, 2022-12-02:
@user897943 Could you share your 3MF to us to reproduce, or just the config? Thank you very much!

user897943, 2022-12-03:
Hi, sorry for the delay, I was out of town.

The file was downloaded from here:
https://thangs.com/designer/Siganberg/3d-model/Bambu%20Cube%20Base%20100mm-106619

I simply used the 0.4, 0.6 and 0.8 profiles on Bambu Studio and nothing else... On the 0.4 I used "Optimal" among the choices.

I used Copymaster PLA from Technology Outlet  (https://technologyoutlet.co.uk/collections/all/products/premium-pla-3d-printer-filament) the colors I used were: Forest Green, Arctic White, Deep Black, and Golden Line. I set up the temps to between 190-220 per the box. The printer options to calibrate everything automatically was ticked. I also ran a auto-calibration before printing.

darkorb, 2022-12-09:
> I consider the printer unfinished until these features are provided. I think it's fantastic as a printer but please do not lock it down so much. Streaming and MQTT are already baked into the firmware, you just need to release a local endpoint for the LAN.

Did you check before posting btw? Not to be rude or anything but MQTT has been being pulled apart on Discord by folks for a fair while. It's even on the default port and doesn't need anything special to connect.

Also on HA (more recent progress) - https://community.home-assistant.io/t/bambu-lab-x1-x1c-mqtt/489510

user897943, 2022-12-10:
> > I consider the printer unfinished until these features are provided. I think it's fantastic as a printer but please do not lock it down so much. Streaming and MQTT are already baked into the firmware, you just need to release a local endpoint for the LAN.
> 
> Did you check before posting btw? Not to be rude or anything @user897943 but MQTT has been being pulled apart on Discord by folks for a fair while. It's even on the default port and doesn't need anything special to connect.
> 
> Also on HA (more recent progress) - https://community.home-assistant.io/t/bambu-lab-x1-x1c-mqtt/489510

Hey :) I tinkered a couple of days after posting and noticed that MQTT is indeed publishing. Hopefully someone smarter than me can figure out all the intricacies and build a HACS plugin for Home Assistant. Would be absolutely amazing to be able to tie the printer status and actions into home automations, I did with Klipper and it made life so much easier. Having the direct camera stream would also open new opportunities for recognizing issues.