bambulab/BambuStudio #722: Improve Version String for automations

b-reich, 2022-11-27:
**Is your feature request related to a problem? Please describe.**
The Version String in the pre build binaries have leading zeros.
eg. `https://github.com/bambulab/BambuStudio/releases/download/v1.3.1.1/Bambu_Studio_win_public-v01.03.01.01-20221126112354.exe` --> `v01.03.01.01`
This makes automation much more complex as necessary, for tools like scoop or homebrew.

**Describe the solution you'd like**
Have the same version in the binary as the git tag.  eg. `Bambu_Studio_win_public-v1.3.1.1-20221126112354.exe` or `Bambu_Studio_win_public-v1.3.1.1.exe`

lanewei120, 2022-12-01:
@b-reich 
Hello

our image is stored in the database, we will use the full version string to query it from DB
simply remove the zero won't work for the ota flow

could you give your scenarios firstly
we can discuss a solution then

b-reich, 2022-12-02:
@lanewei120 makes something like this https://github.com/orgs/Homebrew/discussions/3990#discussion-4605114 unnecessarily complex. (unfortunately not quite finished, but the problem should be recognizable)

lanewei120, 2022-12-02:
@b-reich
understand

currently we removed the leading zeros when generating tags
we will  keep them in tags from next release

thanks

lanewei120, 2023-01-11:
we have changed the tag to the same format [v01.04.01.52](https://github.com/bambulab/BambuStudio/tree/v01.04.01.52) as the filename

close it

closed 2023-01-11 by lanewei120