bambulab/BambuStudio #724: [ISSUE] Unstable Bambu Studio 01.03.01.01 update for Mac

BambuLabDiscord, 2022-11-28:
The recent Recent Bambu Studio 01.03.01.01 update for Mac is causing slow network connections and unstable performance. The print jobs send over the cloud are super slow now, or sometimes fail or hang all together. This was never an issue in the previous versions. I believe in a effort to address that security issue blog Bambu is now breaking the reliability and speed of our jobs. If you look closer at that blog and issue he publicized, you would realize it's a non-issue on local network, especially behind NAT, like 99.9% of us are. Please stop being reactionary because of one loud (wrong) voice.

SaltWei, 2023-02-16:
Still exist on 1.4 version?

SaltWei, 2023-03-02:
v1.5.0 public beta has been realeased. Please let us known if it still exists. Thanks!

closed 2023-03-02 by SaltWei