bambulab/BambuStudio #725: Detect narrow internal solid infill

whiskeywagon, 2022-11-28:
Narrow internal solid layer detection is always on and and some cases causes the pattern to show up on the top layer, which is not ideal.

Solution: Add "Detect narrow internal solid infill" box so you can enable or disable it.

SaltWei, 2022-11-28:
Could you share one model or case for us to test and debug? Thanks you very much!

whiskeywagon, 2022-11-28:
Sure! I added some pictures too in the zip file. So what I mean is that the layer before top layer overlaps a little bit in the middle and so tranforms it to the top layer.

[Model.zip](https://github.com/bambulab/BambuStudio/files/10104562/Model.zip)

SaltWei, 2022-11-29:
My printing is better than yours.
[image]

We guess maybe have relationship with the printer. Could you follow thi guide to clean the xy axis and re-print again to check whether be improved?
https://wiki.bambulab.com/en/x1/troubleshooting/abnormal-xy-axis-resistance

whiskeywagon, 2022-11-29:
I printed two. Upper one is printed "_Detect narrow internal solid infill_" disabled, below is the normal one.
XY axis cleaned with IPA, both prints all default speed settings (standard 0.2, general PLA), PLA (good quality) 220C/55C, textured plate.
[image]

So you can see that the upper print looks better :)

SaltWei, 2022-11-30:
OK. Please wait for next release, and we will open the option.

whiskeywagon, 2022-11-30:
Great!

closed 2022-11-30 by whiskeywagon