bambulab/BambuStudio #726: Update of Studio loses login information

randyre, 2022-11-28:
Have a valid logged in instance of studio installed. Then update to newer version and you have to login again as if it's a new install.

Expected behavior is it should actually upgrade the install, not just remember some settings.

Login information should remain persistent on an "upgrade"

c2h5oh, 2022-11-28:
Confirmed - update logged me out too.

cheshire76, 2022-12-01:
The need for signing in again likely has to do with the change from HTTP to HTTPS. [https://blog.bambulab.com/answering-network-security-concerns/](url)

StoneLiBambulab, 2023-01-09:
Due to the security upgrade, the user needs to log in again. I am very sorry that the inadequacy in security has affected the user experience.

Stone Li

SaltWei, 2023-03-02:
v1.5.0 public beta has been released.

closed 2023-03-02 by SaltWei