bambulab/BambuStudio #728: [Bug] PEI sheet: broken first layer including Bambu own PEI sheet

H4kunaM4tata, 2022-11-29:
How to replicate: Print on a PEI sheet without glue, specially high temp materials like ASA and above.

Before you say it, nope, glue is not release agent and for over 3y I never used glue to print on my PEI sheets.
Only newbies starting now don't bother using glue. 
Those who have been into 3D printing for sometime knows that all you need is a solid bed adhesion, not glue.
Back to the problem....

This is how a first layers looks like on a steel PEI smooth sheet.
That is wrong, You shouldn't be able to see lines!
In some cases you can actually see the second layer due to gaps between those lines.

[image]

This is more likely how a first layer should look like.
Observe how the lines are "squashed" against each other.
This means the nozzle is lower improving the bed adhesion.

[image]

Textured before:

[image]

Textured after:

[image]

ASA (black)  and PETG (red) printed without warping or coming off the bed after the fix

[image]

[image]

More users are having warping issues even with Bambu Lab PEI sheet.

The workaround for the problem for now:

[image]

Scroll all the way down until you find `draw extrinsic para cali paint` and paste the code block:

`;===== for Textured PEI Plate, lower the nozzle by z-0.04 ===============`
`;curr_bed_type={curr_bed_type}`
`{if curr_bed_type=="Textured PEI Plate"}`
`    G29.1 Z-0.04  ;for Textured PEI Plate`
`{endif}`

This will lower the nozzle by `-0.04 `every time Textured sheet is selected.

[image]

Because of that, you must add 0.06-0.08 elephant foot compensation

[image]

This is not an official fix and will be lost during new releases.
I would like to ask the team to review this and make this change permanent with a proper comment to easily identify it.

All the credits for this workaround goes to: https://www.printables.com/model/322000-bambu-lab-modified-start-and-end-gcode

Thanks a lot

aridShrub, 2022-11-29:
[image]

SaltWei, 2022-12-02:
> Scroll all the way down until you find `draw extrinsic para cali paint` and paste the code block ABOVE IT:
> 
> `;===== for Textured PEI Plate, lower the nozzle by z-0.04 ===============` `;curr_bed_type={curr_bed_type}` `{if curr_bed_type=="Textured PEI Plate"}` ` G29.1 Z-0.04 ;for Textured PEI Plate` `{endif}`
> 
> This will lower the nozzle by `-0.04 `every time Textured sheet is selected.
> 
> [image]
> 
> Because of that, you must add 0.06-0.08 elephant foot compensation
> 
> [image]
> 
> This is not an official fix and will be lost during new releases.

This is same gcode with our internal version and it has been included in next user release version which will come soon

H4kunaM4tata, 2022-12-02:
> [image]
> 
> This is PETG - no z-offset->g-code was used. Is this the aesthetic you are after? 
> 
> 

Close enough 

[image]