bambulab/BambuStudio #729: Cold/Hot/PEI Plate Identification in Studio Slicer

rrik, 2022-11-29:
Feature Request: Please allow markings on the plate in bambu studio slicer for quick identification of the plate being printed on. For example, Cold plate = COLD (in blue); Hot plate = HOT (in red); PEI plate = PEI (in gold). These markings will be shown on the plate when slicing items for easy recognition.

user897943, 2022-11-30:
YES! I will also copy a comment I made on reddit:

Edit4: I noticed that each build surface has a ArUco marker, this identifies the type of plate being used. I noticed that if I select PEI plate on the slicer, the bed temp will be adjusted accordingly (PEI needs to be hot to stick). I forgot to amend from Cold Plate to PEI plate and lost two prints because they detached due to the bed being colder than PEI requires.... I find it a little wasteful that the plate marker is read before printing but it does not compare to the slicer settings and throws a warning if they mismatch... For example if my Bambu slicer sent a job that had a Cold plate but the printer scans and sees it's the PEI plate, then it should warn and pause the print before it starts! So that I can rectify the situation. I know I can double check before slicing but I have ADHD and it's hard to remember everything. The printer is smart enough to scan, why not use this to our advantage? In the warning it could even go as far as offering to "change plate to PEI" or whatever it scanned... that way the print can be continued if the filament allows - since each plate is compatible with a number of specific filaments which would have been included in the sliced 3mf.... So the printer should be smart enough to offer the user to continue the print and auto-swap to the installed plate. Or at least let the user manually change the plate and hit continue.

YifanWuBambu, 2022-11-30:
@rrik 
Like this?
[image]

SaltWei, 2023-10-08:
Adding color will make the bad visual effect. It will influence people's attention on the colored models. So finally we only add text.

Thank you very much for good suggestions!

closed 2023-10-08 by SaltWei