bambulab/BambuStudio #731: Font Selection is virtually impossible

megamaniac, 2022-11-29:
The font selection in the slicer is virtually impossible at the moment. I am not sure if it is reading every instance of every font on my system, but there seem to be thousands.

- It scrolls so fast I can't stop on a font
- I can't search for the font with the keyboard (keypresses still effect the objects)
- They don't seem to be in alphabetical order
- If you don't know the exact style of font you want, there is no way to see a preview.

It would be great to see a pop up for the font selection, or some way to customise what fonts are shown. At the moment it is essentially broken.

YifanWuBambu, 2022-11-30:
The font preview will included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu