bambulab/BambuStudio #732: Random or selected multi-color options with no purge in bambu studio.

Eltruvel, 2022-11-29:
It would be awesome to have a random color option with no purge in Bambu Studio, to create a unique rainbow colored print without having the X1 purge material out of the purge shoot during a color change. This is to save on the cost of wasted material.

c2h5oh, 2022-11-30:
This would be a great showcase for AMS and what unique options it brings to the table.

To simplify implementation for you here's a suggested v1 feature set:
- only support identical filaments (same preset, same print settings)
- let users add a sequence of filaments to use in the following format: 
    - use this filament for x meters / use this filament for x layers / use this filament until it runs out
    - purge on/off
    - once done select: next filament on the list / another filament from the list at random

This would allow for stripes of alternating colors, smooth-ish or sharp transitions (multicolor filaments change color every 10-20m - too much for smaller prints). It would also implement another requested feature: ability to use multiple identical spools of filament for larger prints.

ppamment, 2022-12-03:
This would be a nice feature.

As a workaround for now you could create an extra printer profile and strip out all the flushing gcode from the change filament section in settings.
Create a model of stacked cubes that cover the build plate with whatever height you like each, import it when you want to make a rainbow and set it to"modifier", then pick the colours.

It's a bit ugly but once you have the profile and modifier stack to import set up it should only take a minute or so to apply to a print each time