bambulab/BambuStudio #735: "Small perimeters" speed setting (like PrusaSlicer)

barrettj, 2022-11-29:
**Is your feature request related to a problem? Please describe.**
I would like a setting to control the speed when printing small perimeters

**Describe the solution you'd like**
A "Small perimeters" speed setting, like PrusaSlicer has (Under Speed for print moves)

This setting is super important for several models that we print and sell.  It's especially important with how fast the X1 is by default.

ppamment, 2022-12-05:
@barrettj FYI this is already available in SoftFever fork of Bambu Studio if you need it while you wait for it to make it into the official slicer: 

https://github.com/SoftFever/BambuStudio-SoftFever

barrettj, 2022-12-05:
> @barrettj FYI this is already available in SoftFever fork of Bambu Studio if you need it while you wait for it to make it into the official slicer:
> 
> https://github.com/SoftFever/BambuStudio-SoftFever

I’ve indeed been using that the past week and my project has been printing flawlessly (and how I was certain this was the feature I needed after trying several other things) - thanks though!

SaltWei, 2023-04-18:
@QingZhangBambu Please evaluate this feature request

QingZhangBambu, 2023-08-08:
we've opened the "seam gap" and "wipe speed" parameters on our latest version.

man55, 2023-09-02:
Very useful feature, please add it.

QingZhangBambu, 2023-10-16:
it will be added in the next version.

XunZhangBambu, 2023-12-21:
It has been added in 1.8 version

closed 2023-12-21 by QingZhangBambu