bambulab/BambuStudio #736: PLA Fan g-code question

mujeebbasit, 2022-11-30:
I was reviewing why the fan noise was so high while printing PETG on a PEI plate.  I could not find any parameters at the filament or slicer settings to control the filter fan.  It seems to run at full speed for the entire print.  I suspect it is because the bed temperature is so high.  I then went to g-code and found this section:
=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_tool]=="PLA"}
    {if (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}
    M106 P3 S180
    {elsif (bed_temperature[current_extruder] >50)||(bed_temperature_initial_layer[current_extruder] >50)}
    M106 P3 S255
    {endif};Prevent PLA from jamming
{endif}
M106 P2 S100 ; turn on big fan ,to cool down toolhead

This section appears to control the fans to prevent PLA jamming but the last line ("M106 P2...") is outside the if statement there it will run for any filament.  This does not appear to be the desired behavior given the comments around the block.

Thank you for the consideration

mujeebbasit, 2022-11-30:
I wanted to add that I did find where the PETG fan code is located.  It is in the filament specific g-code:
 ; filament start gcode
{if (bed_temperature[current_extruder] >60)||(bed_temperature_initial_layer[current_extruder] >60)}M106 P3 S255
{elsif (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}M106 P3 S180
{endif}

This makes me even more confused about the PLA specific g-code in the main start code.

SaltWei, 2022-11-30:
> but the last line ("M106 P2...") is outside the if statement there it will run for any filament. This does not appear to be the desired behavior given the comments around the block.

This is in start gcode and used for all filament. Because bed level and calibration costs long time. This only works for prepare stage for lower chamber temperature. Not a bug. When model printing is started, this P2 fan is controlled by cooling settings in filament setting page.

SaltWei, 2022-11-30:
More information：
When model printing is started， the part cooling fan(P1) and auxiliary part(P2) cooling fan is controlled by cooling settings in filament setting page. But for the P3 fan, it's absolutely controlled by filament gcode according to the bed temperature.

For PETG, it's two stage(45 and 60), and maybe this is nor reasonable. We will make it higher to  be 60 and 70.
And it's open to users to modify the filament gcode.

mujeebbasit, 2022-11-30:
Thank you for the quick response.  The information is very helpful.  It would be nice to add a comment to that line of code highlight the reason for using the fan.  It is a clever solution but under the section it implies that it is only for PLA.

Regarding the chamber fan, it would be ideal to have it be responsive to the chamber temperature instead of having a static fan speed that cannot be changed during printing.  The actual chamber temperature is dependent on so many factors from ambient temperature to whether the door is open or not.  Having a temperature responsive fan control could significantly reduce noise and improve longevity of the fan.

WeiTaoSJTU, 2022-11-30:
> Regarding the chamber fan, it would be ideal to have it be responsive to the chamber temperature instead of having a static fan speed that cannot be changed during printing. The actual chamber [...]

Yes, you are right. But this need to be developed by printer firmware, and slicer can't do anything.
I asked printer developer about this feature, and they said the accurate chamber temperature can't be got.

SaltWei, 2022-11-30:
The chamber temperature is not accurate, so maybe this feature can't be developed. You can raise this feature request on APP to printer team. https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

SaltWei, 2022-11-30:
> Regarding the chamber fan, it would be ideal to have it be responsive to the chamber temperature instead of having a static fan speed that cannot be changed during printing.

I agree with you, but this is not studio feature request. We need to comfirm with printer firmware team.