bambulab/BambuStudio #737: Could we have snug supports?

Lauris1512, 2022-11-30:
Hello, we really need snug supports to decrease material waste. The current supports can not be made snug and even with support blocker it doesnt work. This makes removing supports on some projects supper chalenging. Solution involves using prusaslicer, but I prefare Bambu studio for the connectivity and default profiles.

Thank you.

user897943, 2022-11-30:
Can you post two photos, one with the way that Bambu prints and one with the way you are suggesting so I can see the difference.

YifanWuBambu, 2022-11-30:
Yes. It was added to our plan list.

randyre, 2022-11-30:
I was just coming to post the same issue.  Not only is it a waste of material, but it can also make the outer surface worse and isn't needed.

You can see the extra material inside the circles I drew.  Also, the box on the lower left shows support material on the "top" where not only isn't it needed, but was also set to no supports. lastly, the arrow shows where supports were placed even though they were supposed to be blocked.

The last photo shows Cura set at 60 degrees and nothing else changed and it comes out exactly as expected without having to use support blockers at all.
[image]
[image]
[image]

randyre, 2022-12-01:
That all being said, in normal mode Cura is almost 10 hours on a highly tuned CR-10S Pro, and its 4.5 hours on the X1.

MusangKingBambulab, 2022-12-03:
Hi guys. We have added a new parameter to adjust the normal support span namely "Normal support expansion" in the coming release. It will bring more space to customize your overhang area of normal support with "Base pattern spacing" jointly. Please check #743.

SaltWei, 2023-03-01:
Available in v1.5.0 Public beta

closed 2023-03-01 by SaltWei