bambulab/BambuStudio #740: [Feature Request] Properly detect and default to hardware OpenGL rendering whenever connected via Remote Desktop Protocol (RDP)

BambuLabDiscord, 2022-11-30:
Bambu Studio is perpetuating an issue from upstream PrusaSlicer (I verified this by running PrusaSlicer as well) where by default it uses the bundled Mesa OpenGL software-based rendering library whenever connected via Remote Desktop Protocol (RDP). This can be problematic on a system with a slow CPU. Since the rendering must then be performed by the CPU rather than the GPU which is often more performant in 3D processing, the 3D plate in the viewport is slow. Even when using a fast CPU like a Ryzen 9 5950X, the 3D viewport rendering is sluggish and practically ususable. The current workaround is to delete or rename the DLL (opengl32.dll) in .../mesa/ so the slicer is forced to call Windows' default system-level hardware OpenGL libraries. Cura (I verified this) does not have this issue like PrusaSlicer or Bambu Studio since the developers of that slicer chose to default to hardware rendering OpenGL via RDP.

hifihedgehog, 2022-11-30:
The issue I found stems to this line:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L255

This comes down to a fundamental misunderstanding by the PrusaSlicer developers of the OpenGL capabilities. RemoteFX does not have to be enabled for hardware OpenGL rendering support to be exposed to the user session environment. While this may have been the case years ago when Remote Desktop Protocol was more limited in feature set, this no longer holds true. [As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy.

hifihedgehog, 2022-11-30:
I did some digging and found the solution. The issue I found seems to stem to this code segment:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L250-L257

The problem is `::GetSystemMetrics(SM_REMOTESESSION)`, which forces software rendering over RDP, which is wrong. It needs to be removed. Here's why.

`! opengl_version_check.load_opengl_dll() || ! opengl_version_check.is_version_greater_or_equal_to(2, 0); ` satisfies the need to check for OpenGL. This part of the code checks the desktop environment to see (1) a system-level OpenGL library is exposed and (2) it reports back whether or not the current hardware operating through that library supports OpenGL 2.0 and above which is required for the 3D viewport. If neither is the case, the Mesa driver will do its job as intended even in an RDP session scenario. Therefore, the code line `::GetSystemMetrics(SM_REMOTESESSION)` is redundant since if the RDP session does not have the previously noted hardware graphics adapter support enabled in Group Policy, it should fail in the final OpenGL version check that I have noted at the end of the cited code segment.

The issue I cited comes down to a fundamental misunderstanding by the PrusaSlicer developers, who are perhaps more versed in Linux/Unix and less so in Windows, of the OpenGL capabilities in RDP. Their mention of RemoteFX in the code comments is baffling as well since their check simply looks to see if the program is running in an RDP session or not. If  is running RDP, that code statement will always enforce software OpenGL rendering via Mesa.

[As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy. Since `::GetSystemMetrics(SM_REMOTESESSION)` impacts all RDP sessions, it does not work. I have verified this. I have a Ryzen 9 5950X and GeForce RTX 3080 system and was testing this today when I had noticed this OpenGL issue while running Bambu Studio. As explained in my initial submission (in the first comment above), I tried PrusaSlicer and Cura afterward as well to see how each handled it. Only Cura handles hardware rendering over RDP correctly.

hifihedgehog, 2022-11-30:
I did some digging and found the solution. The issue I found seems to stem to this code segment:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L250-L257

The problem is `::GetSystemMetrics(SM_REMOTESESSION)`, which forces software rendering over a non-RemoteFX RDP session, which is wrong. It needs to be removed. Here's why.

`! opengl_version_check.load_opengl_dll() || ! opengl_version_check.is_version_greater_or_equal_to(2, 0); ` already satisfies the need to check for proper OpenGL support. This part of the code checks the desktop environment to see (1) a system-level OpenGL library is exposed and (2) it reports back whether or not the current hardware operating through that library supports OpenGL 2.0 and above which is required for the 3D viewport.

If neither is the case, the Mesa driver will do its job as intended even in a non-RemoteFX RDP session scenario. Therefore, the code line `::GetSystemMetrics(SM_REMOTESESSION)` is redundant since if the RDP session does not have the previously noted hardware graphics adapter support enabled in Group Policy, it should fail in the final OpenGL version check that I have noted at the end of the cited code segment.

The issue I cited comes down to a fundamental misunderstanding by the PrusaSlicer developers, who are perhaps more versed in Linux/Unix and less so in Windows, of the OpenGL capabilities in RDP. Their mention of RemoteFX in the code comments is baffling as well since RemoteFX does not necessarily have to be enabled to use enable hardware rendering.

[As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy. I have verified this. I have a Ryzen 9 5950X and GeForce RTX 3080 system configured in just this way in group policy _without_ enabling RemoteFX. I was testing this today when I had noticed this OpenGL issue while running Bambu Studio. As explained in my initial submission (in the first comment above), I tried PrusaSlicer and Cura afterward as well to see how each handled it. Only Cura handles hardware rendering over RDP correctly.

hifihedgehog, 2022-11-30:
I did some digging and found the solution. The issue I found seems to stem to this code segment:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L250-L257

The problem is `::GetSystemMetrics(SM_REMOTESESSION)`, which forces software rendering over a non-RemoteFX RDP session, which is wrong. It needs to be removed. Here's why.

`! opengl_version_check.load_opengl_dll() || ! opengl_version_check.is_version_greater_or_equal_to(2, 0); ` already satisfies the need to check for proper OpenGL support. This part of the code checks the desktop environment to see (1) a system-level OpenGL library is exposed and (2) it reports back whether or not the current hardware operating through that library supports OpenGL 2.0 and above which is required for the 3D viewport.

If neither is the case, the Mesa driver will do its job as intended even in a non-RemoteFX RDP session scenario. Therefore, the code line `::GetSystemMetrics(SM_REMOTESESSION)` is redundant since if the RDP session does not have the previously noted hardware graphics adapter support enabled in Group Policy, Mesa will be flagged to be enabled in the final OpenGL version check that I have noted at the end of the cited code segment.

The issue I cited comes down to a fundamental misunderstanding by the PrusaSlicer developers, who are perhaps more versed in Linux/Unix and less so in Windows, of the OpenGL capabilities in RDP. Their mention of RemoteFX in the code comments is baffling as well since RemoteFX does not necessarily have to be enabled to use enable hardware rendering.

[As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy. I have verified this. I have a Ryzen 9 5950X and GeForce RTX 3080 system configured in just this way in group policy _without_ enabling RemoteFX. I was testing this today when I had noticed this OpenGL issue while running Bambu Studio. As explained in my initial submission (in the first comment above), I tried PrusaSlicer and Cura afterward as well to see how each handled it. Only Cura handles hardware rendering over RDP correctly.

hifihedgehog, 2022-11-30:
I did some digging and found the solution. The issue I found seems to stem to this code segment:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L250-L257

The problem is `::GetSystemMetrics(SM_REMOTESESSION)`, which forces software rendering over an RDP session (except in a case I cite below), which is wrong. It needs to be removed. Here's why.

`! opengl_version_check.load_opengl_dll() || ! opengl_version_check.is_version_greater_or_equal_to(2, 0); ` already satisfies the need to check for proper OpenGL support. This part of the code checks the desktop environment to see (1) a system-level OpenGL library is exposed and (2) it reports back whether or not the current hardware operating through that library supports OpenGL 2.0 and above which is required for the 3D viewport.

If neither is the case, the Mesa driver will do its job as intended. Therefore, the code line `::GetSystemMetrics(SM_REMOTESESSION)` is redundant since if the RDP session does not have the previously noted hardware graphics adapter support enabled in Group Policy, Mesa will be flagged to be enabled in the final OpenGL version check that I have noted at the end of the cited code segment.

The issue I cited comes down to a fundamental misunderstanding by the PrusaSlicer developers, who are perhaps more versed in Linux/Unix and less so in Windows, of the OpenGL capabilities in RDP. Their mention of RemoteFX in the code comments is baffling. The RDP session will only appear as local user session if a RemoteFX **_vGPU_** is enabled. A RemoteFX vGPU (a subset of RemoteFX and not RemoteFX itself) isn't necessary to have the necessary hardware graphics support enabled.

[As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy. I have verified this. I have a Ryzen 9 5950X and GeForce RTX 3080 system configured in just this way in group policy _without_ enabling RemoteFX. I was testing this today when I had noticed this OpenGL issue while running Bambu Studio. As explained in my initial submission (in the first comment above), I tried PrusaSlicer and Cura afterward as well to see how each handled it. Only Cura handles hardware rendering over RDP correctly.

hifihedgehog, 2022-11-30:
I did some digging and found the solution. The issue I found seems to stem to this code segment:

https://github.com/bambulab/BambuStudio/blob/1b07874ea3e1bb418f886df1e0759da23483059a/src/BambuStudio_app_msvc.cpp#L250-L257

The problem is `::GetSystemMetrics(SM_REMOTESESSION)`, which forces software rendering over an RDP session (except in a case I cite below), which is wrong. It needs to be removed. Here's why.

`! opengl_version_check.load_opengl_dll() || ! opengl_version_check.is_version_greater_or_equal_to(2, 0); ` already satisfies the need to check for proper OpenGL support. This part of the code checks the desktop environment to see (1) a system-level OpenGL library is exposed and (2) it reports back whether or not the current hardware operating through that library supports OpenGL 2.0 and above which is required for the 3D viewport.

If neither is the case, the Mesa driver will do its job as intended. Therefore, the code line `::GetSystemMetrics(SM_REMOTESESSION)` is redundant since if the RDP session does not have the previously noted hardware graphics adapter support enabled in Group Policy, Mesa will be flagged to be enabled in the final OpenGL version check that I have noted at the end of the cited code segment.

The issue I cited comes down to a fundamental misunderstanding by the PrusaSlicer developers, who are perhaps more versed in Linux/Unix and less so in Windows, of the OpenGL capabilities in RDP. Their mention of RemoteFX in the code comments is baffling. The RDP session will only appear as local user session if a RemoteFX **_vGPU_** is enabled (see [here](https://stackoverflow.com/a/52823314/2225395) and [here](https://www.delphipraxis.net/1294200-post3.html)). A RemoteFX vGPU (a subset of RemoteFX and not RemoteFX itself) isn't necessary to have the necessary hardware 3D graphics support to run the 3D viewport.

[As explained here at Stack Overflow](https://stackoverflow.com/a/61582432/2225395), Windows simply needs "Use hardware graphics adapters for all Remote Desktop Services sessions" enabled in Group Policy. I have verified this when optimizing for RemtoeFx. I have a Ryzen 9 5950X and GeForce RTX 3080 system configured in just this way in group policy ([see here](https://www.reddit.com/r/sysadmin/comments/fv7d12/pushing_remote_fx_to_its_limits/); RemoteFX with hardware rendering enabled is very good for remote work, by the way!). I was testing this today when I had noticed this OpenGL issue while running Bambu Studio. As explained in my initial submission (in the first comment above), I tried PrusaSlicer and Cura afterward as well to see how each handled it. Only Cura handles hardware rendering over RDP correctly.

closed 2022-12-09 by lanewei120

hifihedgehog, 2025-04-16:
@lanewei120, Bambu Studio has experienced a regression with the new Bambu Studio 2 release. Hardware rendering is no longer performed via RDP and the old code line noted in this issue conversation...
https://github.com/bambulab/BambuStudio/blob/904dc26c24da1df94a18bf51f08a7714c088eb83/src/BambuStudio_app_msvc.cpp#L233
...has been reintroduced. I verified this with an older release as well as OrcaSlicer. Please remove the line above. Thanks!

commit 7e96cbf7 (2022-12-09): Fixes #740

Removing check for local or Remote FX virtualized RDP session. The checks below this line already verify if proper OpenGL support is present.