bambulab/BambuStudio #743: Support blocker does not block. Supports generated anyways

randyre, 2022-12-01:
BL Studio 01.03.01.01   
Windows 10.

Support blocker does not block areas that you don't want to be supported. Areas generate a support structure anyhow.

Also, see #737 as its somewhat related.

[image]
[image]

MusangKingBambulab, 2022-12-02:
Hi @randyre, could you please share the 3mf with me? Let me check what is going on.

randyre, 2022-12-03:
I can't send you the 3mf as it contains a proprietary file that I can't share. I did provide a file with the just overhang area. I hope that helps.  I could send you my settings if we had a way to import and export profiles :)

[Anderson Lower BL test.zip](https://github.com/bambulab/BambuStudio/files/10145257/Anderson.Lower.BL.test.zip)

MusangKingBambulab, 2022-12-03:
@randyre I understand your problem and have been solved in the coming release. My recommonded setting is, erasing all the support painting , generating support under "Normal(auto)"，disabling "Don't support bridges" and shrinking the support span with -1mm with the newly introduced parameter "Normal support expansion". The result shows below:

[image]

randyre, 2022-12-05:
@MusangKingBambulab  Thanks, I guess I'll have to wait for the next release.  We don't have "Normal Support Expansion" in the current release.

randyre, 2022-12-20:
I can't get this to work. I keep making the negative number larger and the support does not change until all of a sudden it just disappears. Can we just get a simple parameter that makes the support flush with the edge of the object it's supporting? This is the normal way every slicer I've ever used does it. Why do we have to have so much extra material?

MusangKingBambulab, 2022-12-21:
Sorry for causing you so much inconvenience. My suggested "Normal support expansion" is -1mm and it will genuinely generate a few extras but not affect too much as shown below. The "snug" mode will be introduced in the next release. Thanks for your suggestion.
[image]

Switch-modder, 2025-04-16:
When will the snug supports be added?

tangketan, 2025-04-23:
Snug supports are already added more than one year ago. 

[image]