bambulab/BambuStudio #745: [Possible Bug] Issues starting print after tall print

TheCableGuy99, 2022-12-01:
**Describe the bug**
This is a possible bug as I'd rather not use my own machine to test it in case I break it so I'll leave that to you guys to test. The issue is that I just printed this (there's two files, it's the main part that attaches to the vacuum named long_reach_vacuum.stl):
https://www.thingiverse.com/thing:5591852

Please note that I printed this stood up so it was a tall print. After it printed the bed was more or less at the bottom of the printer. I continued to print the second part laid down, but when starting the print the printer was making some funny noises. I presumed as I was in the other room on the printer that it was trying to lower the bed before raising it and it was already lowered to it's maximum and therefore putting strain on the bed. When it did raise up it didn't raise all the way, stopped around 8/10 the way, lowered again and then proceeded to raise the final part (possibly not used to the bed being so low and thought it should have touched by now?).

I get the feeling the bed easily forgets it's position and needs to remember where it is so it doesn't try lowering when at the bottom. I could be wrong because as said I'd rather not test this again with my own machine but there were plenty of weird noises going on.

**3mf File for This Bug**
Link provided above

**To Reproduce**
Steps to reproduce the behavior:
1. Slice first part and print stood up.
2. Don't home, slice the second part of the model and observe behaviour when starting the next print.

**Expected behavior**
To raise the bed without lowering when it is so low. To also raise the bed fully on first attempt and not lower and re-raise when nearly at the top. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Latest version

Lesani, 2022-12-02:
I just finished a max-height print, it behaves VERY wrong.
My Printer is running firmware 01.00.00.52 as shipped because there is no SD-Card update firmware function, there is no RJ45 port and I do not have an open wifi and at the work site where this printer is located and I am not allowed to put one there

Video is 3min34s and worth watching till the end
The orange message is 1st order vibration mismatch, I always get this, no matter how often I let it calibrate

https://www.youtube.com/watch?v=8meATTUqJOQ

c2h5oh, 2022-12-03:
This also happens with the latest firmware - I've just seen it this morning starting a new print right after a 200mm tall print.

Lesani, 2022-12-12:
@lanewei120 not necessarily "printer" though, is it... just home the Z axis before you move it blindly down in starting gcode?

lanewei120, 2022-12-12:
From our printer team:
this phenomenon is a corner case we haven't metion . In the start-gcode , when start printing , the printer will lower the Z-axis motor current to about 0.4A , then move the bed down and move it up again . If the bed had already at the bottom , the Z-axis will lost steps ---- which is accepted with 0.4A current in our test ----- and then move up , so the printer will not do the homing calibration at the bottom. But if the bed is stuck at the bottom , then the printer will not be able to move the bed up , and the following calibration will fail . 
So the real problem is that 0.4A current was too low when moving up the bed , the bed cannot move up if there is a bit friction at the bottom . We will fix this by modifying the start-gcode .
for now you can modify it yourself ,  add line "M17 Z0.7" before "G0 Z-6" that is at line 6 in the start gcode
[image]

TheCableGuy99, 2022-12-12:
Added, thank you!

christianjou, 2022-12-12:
> Added, thank you!

how did you do to add it ?

TheCableGuy99, 2022-12-12:
Where you select your printer in the top left of the slicer, click the icon to the right of it to edit it.

On the window that opens select "Machine G-code", in the "Machine start g-code" section add the code as above in the screenshot.

Save under a new profile name like and use that until the fix is officially released then you can go back to the default profile.

LuSEllDen, 2023-02-07:
I really don't know why such big bed movement is needed as a first movement of the printer. It happened to me two times after tall prints that bed wants to go down and I can hear the belt skipping the Z gear :-(
If there are piezo sensors on the bed, than bed can go up instead of down because the hit or increased preasure to the nozzle can be detected without any major damage.