bambulab/BambuStudio #746: Arrange objects places objects on purge lines

Lesani, 2022-12-01:
**Describe the bug**
[image]

Objects placed on purge line, but the color-change tower has LOADS of space around it

**To Reproduce**
Steps to reproduce the behavior:
1. import lots of parts
2. auto-arrange
[image]

**Expected behavior**
avoid purge line and calibration lines, but get as close to them as possible

**Desktop (please complete the following information):**
 - OS: w10
 - Version [image]

tangketan, 2022-12-03:
Avoiding the purge line is easy, I'll add it in quickly.
Getting close to wipe tower is difficult: the exact volume of wipe tower is unknown before slicing. You may have noticed that the wipe tower may be larger to smaller after slicing. So in auto-arranging stage I have to estimate the rough volume. In your case obviously the estimated volume is too large. For now you have to manually check the gcode view AFTER slicing if there is wipe tower. In the future we may use some two-pass arranging strategy to solve this problem.

Lesani, 2022-12-03:
I do, I have printed a lot on MMU2s Prusa i3s.
The way it is done there, is that the wipe tower generally only grows in one direction, and that is indicated already in design stage. then after slicing, the wipe tower size will not change anymore unless you change filament settings like purge volumes, or materials, or add/remove objects from the plate. so after the first slicing the tower is modeled correctly in what you call "Prepare", and can be properly avoided when arranging.
This would be a juge improvement, because - I think - you use that large space for the worst-case option of a 16 color print, and that is not an everyday thing.

The same goes for the whole measuring area in the front, it is very very largely avoided, making the printbed much smaller. Since this doesn't change just because objects are rearranted (on the same bed) it could be modeled and avoided narrowly instead.

tangketan, 2023-04-10:
Yes, what you said can be a solution, but it requires a two-pass procedure: arrange globally, then calculate accurate wipe tower size, and then arrange per-plate again. This will double the arranging time. I'll see how to do it properly.