bambulab/BambuStudio #748: Alternative Tree Supports.

wirlybird, 2022-12-01:
Hi, I find tree supports can be very good for certain types of models.  However, the tree supports in Bambu Studio are more trouble than helpful at times.  They tend to engulf the model in a cocoon with hard to remove supports.

I use the tree supports in Flashforges Flashprint a lot and find them to be very good.
Although the Flashprint supports look similar to resin style supports they are for filament (FDM) printing.

I have attached two pictures.  One is of the Flashprint tree supports and the other is the Bambu Studio tree supports.
The same model was used at the same size for both pictures.
As you can imagine the BambuStudio tree supports would be quite difficult to remove on this model without breaking parts of the model.

[image]
[image]

tangketan, 2023-03-21:
Hi @wirlybird , now we have Tree Slim style for tree support. Have you tried that? Is that good enough for you?

wirlybird, 2023-03-21:
Hi,

I’ve only tried them a little bit so I can’t really report on them!  Thank you though.

closed 2023-08-02 by tangketan