bambulab/BambuStudio #750: Cutting a color painted object removes all paint

alexzogh, 2022-12-02:
**Describe the bug**

I attempted to cut a painted model that was too big for the plate. As soon as I cut the model, all of the paint disappeared on the colored model. 

**3mf File for This Bug**

**To Reproduce**
Steps to reproduce the behavior:
1. choose any painted object in the 3mf
2. Click on the cut icon
3. click on 'perform cut'
4. See error - the object will now be cut, but naked with no paint

**Expected behavior**

I expected the object to be cut, but all of the paint to be intact

**Screenshots**

[image]
Photo of a selected colored object about to be cut

[image]
Photo of the same selected colored object after clicking "perform cut" - all painting has been removed

**Desktop (please complete the following information):**
 - OS: Windows 10 version 22H2
 - Bambu Studio version 01.03.01.01

lanewei120, 2022-12-02:
it is the current behavior by design,  not a bug
marked as a feature

randyre, 2022-12-03:
I'm beginning to wonder why so much of the basic functionality of Prusia Slicer has been cut out in the first place. It seems harder to add all this back in than to just make PS work as is with an X1.

YifanWuBambu, 2022-12-03:
@alexzogh @randyre 
This multi-color model is not generated by the multimaterial painting tool. It's color is assigned by specify the extruder of each part of object in the object list or context menu. Bambu Studio also support this.
[image]

ResiduaLiu, 2022-12-09:
> How in the world is undoing and fundamentally changing everything about an object when you just want it cut a 'feature'? Who in the world would ask for such a feature? When I resize the object I don't lose the painting. When I Split, or modify (add, subtract, etc..) it doesn't lose the painting. So are those all bugs then? Very inconsistent behavior when some features behave one way, and others don't.

You are right, but I think you misunderstand the reply. "Marked as feature" means this github issue will be labeled as "feature" rather than "bug". It is a reference for the developers.  The Bambu team will solve the problem you proposed.

rafffel, 2023-08-28:
Any news about that one ?

randyre, 2023-08-29:
@YifanWuBambu
Is there a way to have it only affect the triangles that touch the cut and put them back to the base color, and leave the rest of the model colored, That way there's not as much work to do to get painted again.

bachrach44, 2023-12-20:
Any update on this? I also find it super annoying to spend all this time coloring an object, only to have all my hard work cancelled when I try to set it up for printing.

remcoder, 2023-12-27:
I just ran into this in v1.8.2.56 too. It's a limitation that's easily encountered.

btw, if you lose the colors accidentally this way. At least know you can CTRL-Z to undo the cut.

partymandkse, 2024-07-04:
I guess this one have been going one for a while, but i'd like to give my +1 to this annoying issue.
Haven't tried Orca slicer yet, but this certainly makes me wannna do it.

pgifford, 2024-08-01:
> I guess this one have been going one for a while, but i'd like to give my +1 to this annoying issue. Haven't tried Orca slicer yet, but this certainly makes me wannna do it.

OrcaSlicer has the same problem which is not surprising since it is periodically updated to the latest Bambu Studio code.

cew214DAL, 2024-12-31:
Adding my +1 as well.  Just ran into the issue when just trying to remove the non-colored edges of a file.  Not touching any of the colored triangles at all.  If making this change is so difficult, how about a feature where you can convert painted objects to actual objects (apologies if that's already a thing and I just missed it).  That way, I can just drag the painted object to another plate, do my cut and resize the painted object(s) to whatever size I need and drag it back to the original plate.

Timefactor, 2025-10-25:
Could we get an update on this issue? Just purchased an X1 Carbon and AMS. Spent time painting a very large shield for a cosplay event and when I went to cut it so it could fit on the 256x256x256 plate, all the painting disappeared. This bug has been open for 3 years now. Doesn't help me gain confidence with the brand when the dev team takes over 3 years to resolve an issue like this.

alexzogh, 2025-10-25:
I was the original author of the bug almost 3 years ago. There is a workflow workaround that works just fine for me, and, once you know it, this won't be an issue going forward.

First, Make your cuts FIRST. Most important, when you make your cuts, make sure you check the box next to "cut to parts"

[image]

When you are done cutting all of your parts, go back to the main section, and click on the icon for "assembly view"

[image]

This will assemble all of the parts into one object, and you can now paint the object when it is assembled, and the paint will be back onto the individual parts.

[image]

Not ideal, but since they added assemblies, I've been doing it this way, and it's worked out well.

CallmeBigfat, 2025-10-25:
> Could we get an update on this issue? Just purchased an X1 Carbon and AMS. Spent time painting a very large shield for a cosplay event and when I went to cut it so it could fit on the 256x256x256 [...]

I have another bug they have not fixed in 3 years so I agree 100%. Its a shame. I have been just putting a negative "slice" for the objects I have worked with after coloring. But I know this is not a solution for everything. I will need to try ops solution above next :)