bambulab/BambuStudio #752: Saved projects files contain no geometry after importing (a/some) 3mf file

Lesani, 2022-12-02:
**Describe the bug**
After importing geometry from the attached prusaslicer mmu_segment2.3mf into a project OR
opening mmu_segment2.3mf in studio, and then saving the file from studio
opening the resulting project file results in:
[image]
[image]

**3mf File for This Bug**
[3mfs.zip](https://github.com/bambulab/BambuStudio/files/10139801/3mfs.zip)
mmu_segment2.3mf -> created with Prusaslicer originally
mmu_segment2_savedinStudio.3mf -> the above file opened, and then re-saved in studio

**To Reproduce**
Steps to reproduce the behavior:
1. open the mmu_segment2.3mf
2. observe a perfectly good file opened in studio, import other files, work with it, arrange things, take about 2-3 hrs to put everything how you want it
3. save the project in studio
4. close studio, come back next day
5. open studio and open your project

**Expected behavior**
Opens the project...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: W10
 - Version 
[image]

Lesani, 2022-12-03:
Opening that file with prusaslicer again, gives a slightly more informative info:
[image]

Lesani, 2022-12-03:
*additional: I re-made my 5 plates, and adapted what needed adapting, exported sliced files, then found the option to "export as generic 3MF" and tried that as well. The same issue occurs, the file is not openable anymore in Studio or other software, same messages as above

lanewei120, 2022-12-07:
@Lesani 
thanks for reporting this issue

this is caused by the model instances not supported in bambu studio anymore
and we have fixed this issue in the new coming version

for above 3mf you provided
I have repaired it and give it here, could you have a check whether it is the one you needed?

[new_3mfs.zip](https://github.com/bambulab/BambuStudio/files/10177324/new_3mfs.zip)

Lesani, 2022-12-07:
Hi! Good to know the issue is fixed, thanks!
As for the file, I re-imported everything and modified it quite a bit - all in one session, so I don't loose progress - and exported the plates as gcode. This here is the file, could you repair this one? The one you did repair is basically just rearranged pieces, this one has all my modifications:
[Rocinante_complete.zip](https://github.com/bambulab/BambuStudio/files/10177788/Rocinante_complete.zip)

lanewei120, 2022-12-08:
> Hi! Good to know the issue is fixed, thanks! As for the file, I re-imported everything and modified it quite a bit - all in one session, so I don't loose progress - and exported the plates as gcode. [...]

please have a try
[Rocinante_complete.zip](https://github.com/bambulab/BambuStudio/files/10181195/Rocinante_complete.zip)

Lesani, 2022-12-09:
Opening works, thanks! It does break again upon re-saving it, but with this file I guess I'll wait for the next version, then it should be fine?

lanewei120, 2022-12-10:
> Opening works, thanks! It does break again upon re-saving it, but with this file I guess I'll wait for the next version, then it should be fine?

yes, the comming version has fixed this issue

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

Lesani, 2022-12-20:
@YifanWuBambu this is NOT fixed. The attached 3mf was saved in the new christmas edition
[image]
and has the same issue
[Rocinante_BambuX1C_5Plate_PLA.zip](https://github.com/bambulab/BambuStudio/files/10269439/Rocinante_BambuX1C_5Plate_PLA.zip)

lanewei120, 2022-12-21:
@Lesani 
please show the detailed steps for how to generate the above 3mf

let me have more check

Lesani, 2022-12-22:
1 Take the 3mf that you fixed: https://github.com/bambulab/BambuStudio/issues/752#issuecomment-1341815309
2 open it in 1.4
3 save it in 1.4
4 close bambu studio
5 open bambu studio
6 open the newly saved file from step 3

lanewei120, 2022-12-23:
this is because the file has already got some errors in it
previously I just repaired the error blocking load
but didn't fix the error will cause saving with incorrect data.

now I repaired these two files totally, which should have no error anymore
[3mf_repaired.zip](https://github.com/bambulab/BambuStudio/files/10292164/3mf_repaired.zip)

please have a try

lanewei120, 2022-12-23:
in Bambu Studio V1.4, we have fixed the issue when using prusa create instances
so if you start from the beginning, it should be ok

however, as you start from the Intermediate file which still has some errors
the issue happens

after also clear the errors in the Intermediate file， there should be no error now

Lesani, 2022-12-30:
Thanks, the fully repaired files seem to work!

uraxlizard, 2024-11-11:
Hello, I have the same problem with my file can you take a look please 
[Uploading NVR-Blister.zip…]()

justingolden21, 2025-01-04:
None of the steps above work for me. I have a valid obj and mtl file on latest bambu studio and it says no geometry