bambulab/BambuStudio #753: Clear seem is created when using spiral vase mode

ppamment, 2022-12-02:
**Describe the bug**

When using spiral vase mode, I am often ending up with clear seams on the resulting print, and when I looked closely at the toolpath, they are also visible there. When observing the printer during printing, you can clearly also see the printer dwell for a moment at that spot, as though it is waiting there and moving z axis. This results in excess filament being deposited along that line and shows a very obvious seam. 

**3mf File for This Bug**
[vase_seam.zip](https://github.com/bambulab/BambuStudio/files/10141930/vase_seam.zip)

**To Reproduce**
Steps to reprod

uce the behavior:

I am using a solid model (a popular one from printables: https://www.printables.com/model/290951-a-slightly-overengineered-olpe-vase

Turned on spiral vase mode. 1 wall, 0 infill, no top layer.

I wondered if the seam was caused by retraction between layers, so tried disabling retraction in the filament overrides, but it gives the same result.

**Expected behavior**
I expect when printing in spiral vase mode for the printer to run in a continuous loop and not leave a seam.

**Screenshots**
Here you can see it in the generated gcode:
[image]

Here you can see it on the actual print:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.03.00.25

QingZhangBambu, 2022-12-12:
i use your 3mf and print it in BambuStudio X1C, the seam is Invisible.
there is the print result
[image]

ppamment, 2022-12-13:
@QingZhangBambu Thanks. Not sure why we are seeing different results but it would be good to figure it out as I'm seeing it on multiple prints.

Please let me know if there is anything else you would like me to provide to help you debug.

For now here is another print where this happened: a purge bin for the printer:
[image]

3mf, step, and a video showing where this artifact is clearly present in the slicer as well as on the model:

[purge_bucket.zip](https://github.com/bambulab/BambuStudio/files/10222372/purge_bucket.zip)

Thanks

QingZhangBambu, 2022-12-15:
@ppamment what filament you used? maybe it matters

SaltWei, 2022-12-16:
For the video you shared, because this model has big shape change and some overhang at the first several layers, we do recommand that don't use too high layer height while using 0.42 line width. If you want to use 0.3 layer height, please incerase the line width as well, to avoid gaps in the first several layers.

For seams issue, we can't reproduce this issue in our printing. But we disable the timelapse when printing.
[image]
Did you enable or disable the timelapse?

SaltWei, 2022-12-16:
Has started one printing again with timelapse enabled. Will update one hour later when it finished

SaltWei, 2022-12-16:
Still can't reproduce this issue. The 3MF I used and video are included in zip. Ignore the first layer which is not very well attached.

[3MF.zip](https://github.com/bambulab/BambuStudio/files/10245120/3MF.zip)
[video.zip](https://github.com/bambulab/BambuStudio/files/10245130/video.zip)
[timelapse.zip](https://github.com/bambulab/BambuStudio/files/10245132/timelapse.zip)

Pay attention that don't use smooth type timelapse in spiral vase mode. This will create seam. We will disable smooth type timelapse in spiral vase mode in next version.

SaltWei, 2022-12-17:
Please see the latest message at https://github.com/bambulab/BambuStudio/issues/846, and update your printer firmware to be latest. This should solve the problem. So this issue is not caused by slicer but old printer firmware. Close

closed 2022-12-17 by SaltWei

gmoissey, 2023-05-15:
I can confirm **that this is not a printer issue but a problem with the slicer**. I'm trying to print with vase mode and a sliced preview has some artifacts on every layer. From what it looks like it is happening because of some tasks that a printer has to do between switching to the next layer. I attache
[image]
[image]
d two screenshots

QuiNz0r, 2023-11-10:
This issue still persists, any ideas? 
[image]

bzwart77, 2023-11-10:
Hello, this problem is real and impacting all vase mode models many are printing on x1 and a P1P.  This issue is becoming more known as seen in these [posts1 ](https://www.reddit.com/r/BambuLab/comments/17qqmr8/vase_mode_vertical_line_appearing_when_using/)and [posts2](https://www.reddit.com/r/BambuLab/comments/11cn3ir/how_do_i_get_rid_of_or_at_least_move_this/)

Xarbenence, 2024-05-09:
I am still getting this issue. I am using Bambu Slicer 1.8.4.51, and a Bambu P1S.

QingZhangBambu, 2024-05-10:
@Xarbenence please update the latest version

QingZhangBambu, 2024-05-10:
[image]
[image]
@Xarbenence this function may help

Xarbenence, 2024-05-15:
Thank you, @QingZhangBambu!