bambulab/BambuStudio #754: Can not print very simple object

theogott, 2022-12-02:
**Describe the bug**
This is a very Simple Object. I tried to print it with different Settings, with PLA or PETG and always run into Issues, see picture.
I also tried to print it with 50% - also did not work.

**3mf File for This Bug**
I have attached the sliced file in a zip-file.

**To Reproduce**
Simply try to print this file.

**Expected behavior**
See Pictures below. I guess it will never work with the PEI Plate, but it also does not work with the cold Plate, and it should work.
Because this is not a difficult Object.

[image]

**Screenshots**
[image]

[image]

[image]

If applicable, add screenshots to help explain your problem.

**Desktop (please comple
[TREX 2v2.zip](https://github.com/bambulab/BambuStudio/files/10143965/TREX.2v2.zip)
te the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

SaltWei, 2022-12-03:
Thank you very much to report issue to make slicer keep improved!
But we can't reproduce this issue. And this doesn't seem to be slicer issue.
Please follow this guide to report issue to printer team to analyse.
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

SaltWei, 2022-12-03:
By the way, is other model printed well by your printer?

theogott, 2022-12-03:
Its strange. I have now switched to red PLA and printing this:
And it works fine.
[image]
[image]

theogott, 2022-12-03:
If you can print the otehr model, then please close the case. It must be a Printer issue (maybe with the Nozzle or the Filament ?) in this case.

SaltWei, 2022-12-03:
We printed the 3MF you shared in zip and look fine. PLA and PETG is used both.
[image]

What brand of PLA and PETG did you use?

theogott, 2022-12-03:
The Problem "Cerprise" PETG ... got possibly too much moisture?
Because it worked very well before.

[image]

I will try to dry it and try again.

SaltWei, 2022-12-03:
> The Problem "Cerprise" PETG ... got possibly too much moisture?
> Because it worked very well before.

We used polylite PETG and have never used this brand.
Will order and try.
From my personal experience, different filament may have some difference, but should not be so large. Usually, limiting the volumetric speed of filament setting can improve.

theogott, 2022-12-03:
I will give it a try. Thanks for your support. 
We can close this case.

closed 2022-12-05 by SaltWei