bambulab/BambuStudio #755: Add bottom surface speed into settings

timothyzalusky, 2022-12-02:
The speed of the bottom surface which is not bridging (ie, when using support material) is not currently controllable and I can't figure out how to influence it in the current settings.  It should be placed beneath top surface speed in the speed tab.  

This is critical when using support material to not have the material fuse too well to the support material.

SaltWei, 2022-12-05:
When support top z distance is not zero, the surface speed is still controlled by bridge speed.
When support top z distance is zero, then the surface is not bridge and its speed is controlled by first layer infill speed.

timothyzalusky, 2022-12-06:
Ahh, I did not see that, may in the future want the capability to have a non-first layer control of infill speed for the reason of breakaway support material, but I am not sure how critical that is.  I think this will be sufficient for my needs.

SaltWei, 2022-12-07:
> I think this will be sufficient for my needs.

OK, please let us known if has some new suggestions. Thank you very much!
Close this issue at the moment.

closed 2022-12-07 by SaltWei