bambulab/BambuStudio #756: Models don't lay perfectly flat when imported.

randyre, 2022-12-03:
**Describe the bug**
bambuStudio does not properly drop model onto build plate.

**3mf File for This Bug**
Included STL file, Cura's .3mf, and a picture of both side by side in studio sliced.

**Steps to reproduce the behavior:**
1. Import the model and slice with supports (any type of support) when sliced with supports some of the model does not have supports where it should be.

2. Import the same model into Cura. No need to do anything, just import. Cura places the file on the build plate.

3. Now export that model from Cura and save as .3mf to get new orientation of model

4. Import .3mf into Studio and studio now does not have to adjust model as cura has already done it, and supports are now generated properly.

**Expected behavior**
Should properly level the model on the build plate

**Screenshots**
No rotation was done within either slicer. The problem seems to be Studios' ability to properly lay the model onto the bed,

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Bambustudio 01.03.01.01
[image]
[SkullCandy files.zip](https://github.com/bambulab/BambuStudio/files/10145211/SkullCandy.files.zip)

Benderrodrigu3z, 2022-12-05:
here's a hint, don't use Cura, or export it from Cura but not as a 3mf and do your edits in the BS Slicer

randyre, 2022-12-06:
@Benderrodrigu3z 

I didn't use Cura initially, I loaded directly from the STL file. Re-read the bug report and you will see Cura was only a troubleshooting step.  I included the file and if you're curious you're welcome to try it and you'll see what the problem is. 

I used Cura to proof the problem and then used Cura as a workaround to get Studio to properly position the model. For some reason, Studio would incorrectly lay the model on the print bed, but Cura did it right the first time, and after exporting as a 3mf I was able to get Studio to detect iot properly too.

The problem is clearly in Studio and how it detects the flat surfaces of some models.

Benderrodrigu3z, 2022-12-06:
Good times when you hit unsubscribe and this nucklehead tags you Sincerely,Shane FSent from my iPhone13

randyre, 2022-12-07:
@Benderrodrigu3z 

Sorry, I had no way to know you unsubscribed, figured you left a comment, you wanted an answer.

Sincerely,  **_this nucklehead_**

ResiduaLiu, 2022-12-09:
Hi Randyre,
Thanks for reporting this issue. The STL file from Cura has some non-manifold faces. That's why they have different slices and support results. (See the defects caused by non-manifold faces below.)
It should be a support generation bug rather than an orientation problem. After repairing the model from Cura, they have the same results. 
For a quick fix, you can use the support painting tool. Select the third type "fill", then left click on the areas where you want to generate supportst-click on the areas where you want to generate supports. (See the last picture.)
Our team will fix it ASAP.

[image]
[image]
[image]

tangketan, 2023-03-07:
It's a bug of small overhang detection. I'll fix it soon.

tangketan, 2023-05-16:
This bug has been fixed. Please wait for our next version V1.7.

SaltWei, 2023-08-23:
Please reopen if 1.7.4 doesn't fix

closed 2023-08-23 by SaltWei