bambulab/BambuStudio #759: External OAuth flow in embedded browser

ptyork, 2022-12-03:
On Windows, when logging in to Bambu Studio, the entire process takes place inside of an embedded browser component. This is fine for a BambuLab login. However, it is not a recommended practice for external, OAuth logins. Instead, it is recommended that apps open an external browser for this workflow. This greatly improves the certainty that the end user has when typing in their credentials and reduces the perceived risk profile of the application.

The relevant IETF RFC is 8252: [https://www.rfc-editor.org/rfc/rfc8252](https://www.rfc-editor.org/rfc/rfc8252)

Of particular note is section 8.12:

```
8.12.  Embedded User-Agents

   Section 9 of OAuth 2.0 [RFC6749] documents two approaches for native
   apps to interact with the authorization endpoint.  This best current
   practice requires that native apps MUST NOT use embedded user-agents
   to perform authorization requests and allows that authorization
   endpoints MAY take steps to detect and block authorization requests
   in embedded user-agents.  The security considerations for these
   requirements are detailed herein.

   Embedded user-agents are an alternative method for authorizing native
   apps.  These embedded user-agents are unsafe for use by third parties
   to the authorization server by definition, as the app that hosts the
   embedded user-agent can access the user's full authentication
   credential, not just the OAuth authorization grant that was intended
   for the app.

   In typical web-view-based implementations of embedded user-agents,
   the host application can record every keystroke entered in the login
   form to capture usernames and passwords, automatically submit forms
   to bypass user consent, and copy session cookies and use them to
   perform authenticated actions as the user.

   Even when used by trusted apps belonging to the same party as the
   authorization server, embedded user-agents violate the principle of
   least privilege by having access to more powerful credentials than
   they need, potentially increasing the attack surface.

   Encouraging users to enter credentials in an embedded user-agent
   without the usual address bar and visible certificate validation
   features that browsers have makes it impossible for the user to know
   if they are signing in to the legitimate site; even when they are, it
   trains them that it's OK to enter credentials without validating the
   site first.

   Aside from the security concerns, embedded user-agents do not share
   the authentication state with other apps or the browser, requiring
   the user to log in for every authorization request, which is often
   considered an inferior user experience.
```
Google also specifies this as the expected workflow in their documentation and examples: [https://developers.google.com/identity/protocols/oauth2/native-app](https://developers.google.com/identity/protocols/oauth2/native-app)

It would be best for Bambu Studio to use best practices here. Though launching an external browser for the OAuth flow is slightly less elegant, it should be no more difficult and would significantly improve the perceived risk that users must face when logging in using any of your external authentication options.

StoneLiBambulab, 2022-12-07:
Hi @ptyork ,
     Thank you very much for your advice.  We plan to improve it according to this process. Could you please help to review our scheme ? The login process is as follows:

[image]

Thanks!

Stone Li

ptyork, 2022-12-07:
Thanks for being receptive and responsive to this!!

So I'm assuming your goal with login is simply to link your external OAuth credentials with a BambuLabs account. I.e., you store the user's identity token in the user's Bambu account record so that you can match a Google-supplied identity with a Bambu one. And that Bambu Studio ONLY needs the BambuLabs identity token (used to communicate with Bambu hosted services).

Assuming this, then your identity management process doesn't need to change too much at all from what you already have in place. I just think rather than launching the login process in the embedded dialog, I'd do the entire thing in a the default web browser. At least that would seem the easiest solution.

So I think if I were to do this, I would:

1) User clicks Login button in Studio...create a simple modal dialog saying something like "Logging in Using Browser" and have a Cancel button...user should never actually see this dialog, but it gives them the ability to cancel the process if they don't complete the sign in process.
2) Retrieve the URL to the web page you currently load in your embedded browser
2a) also find an available port on localhost
3) Launch the default browser with the URL (append a parameter with the port number from step 2a)
- note that I think you need only call Win32's ShellExecute with the URL to launch and open the URL in the default browser...no need to muck around in the registry
4) Still in Studio, open a TCP socket to listen on the loopback adapter (on the port from step 2a...i.e., 127.0.0.1:<some port>). Block BambuStudio here waiting for a response or a Cancel button click.
5) Now on the web server Process the login as you currently do. Presumably match external token to bambu identity. This should all take place in the default browser while Studio waits patiently with the dialog open.
6) When the process completes (user is logged in or cancels), the web server should respond with an HTTP redirect to 127.0.0.1:<some port> from #2a, adding in URL parameters that contain a status code and the BAMBU identity token (assuming a successful login).
7) Browser redirects and opens this "page", which is really BambuStudio's open socket. Harvest the token and respond with an HTTP redirect to a page that indicates a successful login and ability for the user to close the browser tab or window.
8) Back in BambuStudio, close the dialog and (if successful) change state to show as logged in.

This is loosely what is in Google's super-simple C# example here: https://github.com/googlesamples/oauth-apps-for-windows/blob/master/OAuthConsoleApp/OAuthConsoleApp/Program.cs

Not sure if that makes sense. But I think it's the simplest solution. Should require only a very small amount of client and server-side work.

Alternatively you could try to do a hybrid of the above by loading the current login page in the embedded browser and then kicking off the external browser only for external logins. That would seem to require more work, but would be a bit more elegant if you believe most users will create and work solely with Bambu credentials.

Hope this helps a little...

StoneLiBambulab, 2022-12-07:
Hi @ptyork ,
     Thank you very much for your patient reply!

Your assuming is right, Bambu Studio ONLY needs the BambuLabs identity token. And listen on the loopback adapter is a more simple way for the response and more friendly for multi-platform. We will follow the workflow as you said.

But there's no enouph time for the latest version, we plan it in the Juanuary version.

Thanks again for your professional reply!

Stone Li

SaltWei, 2023-03-01:
Available in v1.5.0 Public beta