bambulab/BambuStudio #760: BambuStudio crashes after login

rnovacek, 2022-12-03:
**Describe the bug**
BambuStudio loads fine and seems to work completely ok until I login to my bambu account. Downloading the network plugin is fine.

I'm using linux version (built using Containerfile) on Fedora 36.

**3mf File for This Bug**
Not related to slicing, works fine when not logged into Bambu account.

**To Reproduce**
Steps to reproduce the behavior:
1. Open BambuStudio
2. Download networking plugin
3. Login into existing Bambu account
4. Whole BambuStudio crashes

**Expected behavior**
No crash :)

**Screenshots**
I tried running it through `gdb` and it seems to crash in the networking plugin:

```
#0  0x00007ffff1e9d0ae in __GI___libc_free (mem=0x1) at malloc.c:3363
#1  0x00007fffee161c67 in err_clear (deall=0, i=0, es=0x7ffdfc003ae0) at crypto/err/err_local.h:86
#2  ERR_clear_error () at crypto/err/err.c:345
#3  0x00007fffcaa1b896 in SSLSocket_close () at /home/radek/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so
#4  0x00007fffcaa135d9 in MQTTAsync_closeOnly () at /home/radek/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so
#5  0x00007fffcaa15aa0 in nextOrClose () at /home/radek/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so
#6  0x00007fffcaa167dd in MQTTAsync_receiveThread () at /home/radek/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so
#7  0x00007ffff1e8cded in start_thread (arg=<optimized out>) at pthread_create.c:442
#8  0x00007ffff1f12370 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

`debug_*.log.0`: https://paste.centos.org/view/d0259b54
`debug_network_*.log` is empty (zero bytes)

**Desktop (please complete the following information):**
 - OS: Fedora 36 with Plasma Desktop

lanewei120, 2022-12-05:
> 

it seems the logfile can bot be accessed
[image]

could you upload it here directly?

rnovacek, 2022-12-05:
Sorry, reuploaded log: [debug_Mon_Dec_05_08_22_29_128843.log.0.txt](https://github.com/bambulab/BambuStudio/files/10151574/debug_Mon_Dec_05_08_22_29_128843.log.0.txt)

The architecture is x64_64:
> file /opt/bambu/bin/bambu-studio 
```
/opt/bambu/bin/bambu-studio: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=5f95f295bb374a8ef37d56a771f2c3396c055488, for GNU/Linux 3.2.0, not stripped
```

> file ~/.config/BambuStudio/plugins/libbambu_networking.so 
```
/home/radek/.config/BambuStudio/plugins/libbambu_networking.so: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=0539e80c8a38e73d34aaff25e67cb378996dbcda, not stripped
```

Benderrodrigu3z, 2022-12-05:
here's a hint, don't use linux..... did you try the Mac version? I ask because its Unix which is technically linux and should work

lanewei120, 2022-12-07:
> here's a hint, don't use linux..... did you try the Mac version? I ask because its Unix which is technically linux and should work

did you build your bambu studio by yourself?
which 'CMAKE_BUILD_TYPE' are you using?Release or Debbug

lanewei120, 2022-12-13:
@rnovacek 

we have updated a new version of libbambu_networking.so(1.3.1.3)
could you remove the old plugins manually(rm -rf /home/user.name/.config/BambuStudio/plugins/*)
and download this new library for a try?

thanks

andrewda, 2022-12-13:
@lanewei120 I have pretty much the same setup as @rnovacek (Fedora 37), and am running into the same crash. Deleting `~/.config/BambuStudio/plugins/*` and re-running did not fix the problem, unfortunately. I confirmed that the new 1.3.1.3 version was installed:

```
2022-12-12 22:52:44.279249[Thread 0x00007efc797fe6c0]:[download_plugin 1]: get type slicer/plugins/cloud, version 1.3.1.3, url https://public-cdn.bambulab.com/upgrade/studio/plugins/01.03.01.03/linux_01.03.01.03.zip
```

lanewei120, 2022-12-13:
> @lanewei120 I have pretty much the same setup as @rnovacek (Fedora 37), and am running into the same crash. Deleting `~/.config/BambuStudio/plugins/*` and re-running did not fix the problem, [...]

it seems you also use Fedora, and maybe it is related with the linux environment;
currently I only have an Ubuntu desktop here, and it works fine on my machine
I will try to find another machine using Fedora to reproduce this issue

rnovacek, 2022-12-13:
It is the same with the new version unfortunately, this is the traceback:

```
#0  0x00007ffff1e9d0ae in __GI___libc_free (mem=0x1) at malloc.c:3363
#1  0x00007fffee161c67 in err_clear (deall=0, i=0, es=0x7ffdd8003c80) at crypto/err/err_local.h:86
#2  ERR_clear_error () at crypto/err/err.c:345
#3  0x00007fffd281cf4c in SSLSocket_close () at /home/radek/.config/BambuStudio/plugins/libbambu_networking.so
#4  0x00007fffd2814855 in MQTTAsync_closeOnly () at /home/radek/.config/BambuStudio/plugins/libbambu_networking.so
#5  0x00007fffd2816ef5 in nextOrClose () at /home/radek/.config/BambuStudio/plugins/libbambu_networking.so
#6  0x00007fffd2817bcf in MQTTAsync_receiveThread () at /home/radek/.config/BambuStudio/plugins/libbambu_networking.so
#7  0x00007ffff1e8cded in start_thread (arg=<optimized out>) at pthread_create.c:442
#8  0x00007ffff1f12370 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

The stdout says I'm using the new version:

```
[2022-12-13 09:55:38.764] [console] [info] bambu_network_get_version, version: 01.03.01.03
```

Could be related to some SSL libraries - maybe different version on Fedora than Ubuntu:

```
> openssl version
OpenSSL 3.0.5 5 Jul 2022 (Library: OpenSSL 3.0.5 5 Jul 2022)

> rpm -q openssl
openssl-3.0.5-2.fc36.x86_64
```

lanewei120, 2022-12-16:
I just installed a Fedora 36
and built a bambustudio from newest codes from github
the login functions work well

[image]

and I Followed the instructions on the [linux build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)

Could you have a try again
and if still failed, then we need to check the difference

andrewda, 2022-12-16:
Following that build guide worked for me on Fedora 37, thanks! There is a slight typo, though: in the environment setup section, `libxkbcommon-dev` should be `libxkbcommon-devel`.

I was previously trying to use the Podman `Containerfile` to build an AppImage, and that still appears to be giving me the same issue. Now that I have something working, I might try to debug that a little more. Thanks again for the help!

lty1993, 2022-12-27:
Same behavior on Archlinux. 

```
Thread 48 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff597fa6c0 (LWP 671024)]
0x00007ffff1dc3b4e in free () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff1dc3b4e in free () at /usr/lib/libc.so.6
#1  0x00007fffe42bed58 in ERR_clear_error () at /usr/lib/libcrypto.so.3
#2  0x00007fffc8cae056 in SSLSocket_close () at /home/lty1993/.config/BambuStudio/plugins/libbambu_networking.so
#3  0x00007fffc8ca5d99 in MQTTAsync_closeOnly () at /home/lty1993/.config/BambuStudio/plugins/libbambu_networking.so
#4  0x00007fffc8ca8260 in nextOrClose () at /home/lty1993/.config/BambuStudio/plugins/libbambu_networking.so
#5  0x00007fffc8ca8f9d in MQTTAsync_receiveThread ()
    at /home/lty1993/.config/BambuStudio/plugins/libbambu_networking.so
#6  0x00007ffff1db38fd in  () at /usr/lib/libc.so.6
#7  0x00007ffff1e35a60 in  () at /usr/lib/libc.so.6
```

rnovacek, 2023-01-19:
Building BambuStudio using `BuildLinux.sh` locally (no podman) partially fixes the issue. The AppImage I got using local build have the connection to printer working and I can even see the printer camera. I'm not able to start a print from BambuStudio though, it shows this error and the print job doesn't start:

[image]

Debug log [debug.log.txt](https://github.com/bambulab/BambuStudio/files/10456659/debug.log.txt), network log is empty.

spuiuk, 2023-01-19:
I've had the same issue running the Bambulab Appimage on my Fedora 37 desktop.

The Appimage crashes with the following output.
```
** (bambu-studio:180848): CRITICAL **: 12:05:42.589: Cannot register URI scheme wxfs more than once

** (bambu-studio:180848): CRITICAL **: 12:05:42.589: Cannot register URI scheme memory more than once
Segmentation fault (core dumped)
```
I have generated a new Appimage using a F36 base image in the container build. This works for me and I can now login to the cloud.

 The containerfile is available  [here](https://github.com/spuiuk/BambuStudio/blob/Container_fedora/Containerfile.fedora)

Copy the Containerfile.fedora to the BambuStudio folder. 

For this build, I mount my current git repo as a volume within the container.  To build, run the following command
```
podman build . -v BAMBULAB_REPO_FOLDER:/BambuStudio:Z  -f Containerfile.fedora

```
The resulting AppImage is available at build/BambuStudio_fedora64.AppImage with the BambuStudio repo.

spuiuk, 2023-01-19:
Even though I can now login to the cloud, attempting to send a print results in the same failure reported earlier by @rnovacek.
[image]

lanewei120, 2023-01-20:
which code are you using?
the newest on the master branch, or the tag [v01.04.02.13](https://github.com/bambulab/BambuStudio/tree/v01.04.02.13)
also what is your linux os version?
@spuiuk

spuiuk, 2023-01-20:
I am on the latest commit of the master branch.
7e9ccefc (FIX: boost: fix the patch failed issue of boost, 2023-01-17)

I am running Fedora 37 on x86_64 architecture. All packages have been updated to the latest. 
 
[I compiled the BambuStudio using a container image with a Fedora base image instead of the original Ubuntu](https://github.com/spuiuk/BambuStudio/blob/Container_fedora/Containerfile.fedora). The steps followed were the same as the[ linux build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide).

I've run it both as app image as well as the executable directly.
With the Fedora based container, I can not connect to the printer but when attempting to upload the print job to the printer, I get the error mentioned. 

Since uploading my Containerfile, I've updated it to use F37 base image to build the package with the same results.

lanewei120, 2023-01-20:
we can reproduce this error here on My Fedora
and we find that the newest codes on github/master has some inconsistencies with the network plugins downloaded

as we just release V01.04.02
Could you checkout to the tag [v01.04.02.13](https://github.com/bambulab/BambuStudio/releases/tag/v01.04.02.13)
and use this one for building?

I just confirmed use this tag is ok here

sorry about this

spuiuk, 2023-01-20:
Thanks @lanewei120. I have started the build but hit a hiccup. 

The [tag](https://github.com/bambulab/BambuStudio/tree/v01.04.02.13/deps/Boost) does not have the file /BambuStudio/deps/0001-Boost-fix.patch which causes the build deps to fail. I have copied this file over from the master branch and that has let me continue with the build.

spuiuk, 2023-01-20:
@lanewei120 The build was successful and I can upload prints to the printer through the cloud using the resulting Appimage.

I build the code using the repo at
https://github.com/spuiuk/BambuStudio/tree/fed_v01.04.02.13
The changes I made are
1) Add a Containerfile.fedora and build using this containerfile.(Used mybuild.sh helper script).
2) I had to copy over BambuStudio/Boost/0001-Boost-fix.patch from the master branch to build the deps.

I have uploaded the Appimage to 
[https://www.spui.uk/BambuStudio/BambuStudio_fedora64_v01.04.02.13.AppImage](https://www.spui.uk/BambuStudio/BambuStudio_fedora64_v01.04.02.13.AppImage) 
if someone else would like to test.

I currently have my printer printing out a test print I sent over the cloud.

jasoncouture, 2023-01-20:
@spuiuk testing now, on Fedora 37

I'll update with how it goes.

jasoncouture, 2023-01-21:
Yeah, camera issue was definitely a printer issue. Booted it from wifi and reconnected ( <3 Unifi ), and it works again. Good work @spuiuk

spuiuk, 2023-01-25:
Thank you. I have compiled from  v1.4.2 branch and can confirm it builds fine. The build is available at
https://www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-4-g261058d1.AppImage

PRouzeau, 2023-01-28:
> Thank you. I have compiled from v1.4.2 branch and can confirm it builds fine. The build is available at https://www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-4-g261058d1.AppImage

Hello, Ubuntu 22.04.01 on my machine, this build and the former one work for me, send file on printer and view camera, though Lan mode was not working (tested on former build only). Only in 720p. 
I have a strange bug: Each time I run the appimage, when I start the camera (and only then) the camera windows became larger and now the right border goes over the right of my screen. What is extremely curious is when trying the new build, it did not reset the window width which continue to increase, so that may came from the camera plugin ? 
Also it refuses to send the file on the SD card, telling me that the printer is not on the same network (true - for safety, no point being in my network if lan mode does not work), meaning the file is not sent via the cloud as the normal print is ?

lanewei120, 2023-01-29:
> e file on the SD card, telling me that the printer is not on the same network (true - for safety, no point being in my network if lan mode does not work), mea

are you using a X1 or P1P?

for the first issue, it maybe a video size issue, we will fix it in the future.

PRouzeau, 2023-01-29:
> are you using a X1 or P1P?
I am using a recent (received 4 days ago) X1 Carbon, only with Linux

lanewei120, 2023-01-29:
"Also it refuses to send the file on the SD card, telling me that the printer is not on the same network (true - for safety, no point being in my network if lan mode does not work), meaning the file is not sent via the cloud as the normal print is ?"

this is caused by the ip address not known to Bambu Studio
we have added a popup window, which will alow you to input the ip and access code manually
you can input the ip there(only for first time)

jasoncouture, 2023-02-03:
I built and installed the latest master @ ad9eccc5
Network send, and print functions fail with an error before uploading the gcode:
[image]

After this error, I get dropped back to print/send screen at 100%, that does not end until I hit cancel.
[image]

Fedora 37:
`6.1.8-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Jan 24 20:32:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

lanewei120, 2023-02-03:
> I built and installed the latest master @ [ad9eccc](https://github.com/bambulab/BambuStudio/commit/ad9eccc5f47d5cc7b7a67d3a1b24f8ff6bc9bee9) Network send, and print functions fail with an error before uploading the gcode: [image]
> 
> After this error, I get dropped back to print/send screen at 100%, that does not end until I hit cancel. [image]
> 
> Fedora 37: `6.1.8-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Jan 24 20:32:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux` [image]

@jasoncouture 
this is the known issue on the 'master' branch
please use the codes on branch 'v1.4.2' currently

master branch has the newest codes, but not stable
we will fix it later

lanewei120, 2023-02-03:
> 

@jasoncouture 
for 1.4.2
1. when it crashes, does the print task sent to printer successfully?
2. is there any crash stack?

I had printed serveral times using 1.4.2 yesterday, and can not reproduce this issue

spuiuk, 2023-02-03:
@jasoncouture, I found that the appimage build based on the ubuntu base image does not work very well on my Fedora 37 workstation  ie. It used to crash when I attempt to login to the cloud. So I had to build my own image using Fedora as the base image. I have been using these appimages on my system and they seem to work very well. Although from #1189, the latest patches may cause issues when viewing the video window for a P1P  while running on Ubuntu.

If you are using Fedora or a similar system, you could try the Appimage I am using to see if that fixes your issue. The latest AppImage I use is available at 
https://www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-14-gd1b93c16.AppImage
This was built from the v1.4.2 branch last evening. I just printed out an object this morning.

lanewei120, 2023-02-03:
double confirmed on branch 1.4.2
the app will crash when "send to sdcard ", and work well when "print"

I just found an issue in the network plugin, and It may cause the crash when send to sdcard
please close the application
then delete ~/.config/BambuStudio/plugins/*.so
and then restart the application to let the new network plugins downloaded

then "send to sdcard" should be ok now

jasoncouture, 2023-02-03:
> @jasoncouture, I found that the appimage build based on the ubuntu base image does not work very well on my Fedora 37 workstation ie. It used to crash when I attempt to login to the cloud. So I had [...]

I was using your build. It worked pretty well, aside from a segfault after sending/printing the print to/on the printer. I built my own based on latest master to see if it was fixed.

jasoncouture, 2023-02-03:
> > 
> 
> @jasoncouture for 1.4.2
> 
> 1. when it crashes, does the print task sent to printer successfully?
Yes. The crash happens after the countdown for switching to the device tab. It also happens sometimes even if I cancel after the print starts.
Happens about 5-10% of the time.

> 2. is there any crash stack?
There is not, just a segfault.
> 
> I had printed serveral times using 1.4.2 yesterday, and can not reproduce this issue

It's random, most of the time the first send/print works ok. But not always.

Additional prints/sends however have a much higher failure rate (~80%)

jasoncouture, 2023-02-03:
> double confirmed on branch 1.4.2 the app will crash when "send to sdcard ", and work well when "print"
>
> I just found an issue in the network plugin, and It may cause the crash when send to sdcard [...]

I'll give it a shot, this also impacted print as well though. I can screenrecord if it happens again after updating the plugin.

jasoncouture, 2023-02-04:
@lanewei120 
Here's a video of the segfault. It's not 100% of the time, and usually does not happen on the first print.
In this case, I printed another project before this one.

It may look like it didn't crash, but in this case I had two instances open (Note, normally I do not have two open, this issue occurs regardless)

https://drive.google.com/file/d/1Q74vujcZ520Ta9ceE6l1v4FwhI0r056C/view?usp=sharing

lanewei120, 2023-02-05:
@jasoncouture 
could you give a snapshot of the directory ~/.config/BambuStudio/plugins
please  use 'ls -l' in that directory and give the result here

I can check whether the network plugins is the new one or not

thanks

jasoncouture, 2023-02-05:
@lanewei120  Sure, here you go
```
[jasonc@holopoint plugins]$ pwd
/home/jasonc/.config/BambuStudio/plugins
[jasonc@holopoint plugins]$ ls -la
total 37368
drwxr-xr-x. 1 jasonc jasonc       90 Feb  4 15:01 .
drwxr-xr-x. 1 jasonc jasonc      130 Feb  5 11:29 ..
drwxr-xr-x. 1 jasonc jasonc       78 Feb  4 15:01 backup
-rw-r--r--. 1 jasonc jasonc 33638760 Feb  3 20:17 libbambu_networking.so
-rw-r--r--. 1 jasonc jasonc  4624352 Feb  3 20:17 libBambuSource.so
[jasonc@holopoint plugins]$ strings libbambu_networking.so  | grep '1\.0' | head -n 2
bambu_network_agent/01.04.02.02
01.04.02.02
```

quantalume, 2023-02-06:
> If you are using Fedora or a similar system, you could try the Appimage I am using to see if that fixes your issue. The latest AppImage I use is available at [...]

This is also working on openSUSE Tumbleweed. I had some issues with the different versions of webkit2gtk available through the package manager, but after keeping libwebkit2gtk-4_0-37 and removing the rest, it works fine.

lanewei120, 2023-02-06:
@jasoncouture 
the network plugin is the newest one

I will try to reproduce it and have a debug firstly
Do you know how to get the callstack with gdb?
if you can get a callstack, it will be great help to us

jasoncouture, 2023-02-06:
```
Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000000002ed0d6b in wxWebRequestImpl::ProcessStateEvent(wxWebRequest::State, wxString const&) ()
Missing separate debuginfos, use: dnf debuginfo-install llvm-libs-15.0.7-1.fc37.x86_64 webkit2gtk4.0-2.38.4-1.fc37.x86_64
(gdb) where
#0  0x0000000002ed0d6b in wxWebRequestImpl::ProcessStateEvent(wxWebRequest::State, wxString const&) ()
#1  0x00000000032e2ad0 in wxEvtHandler::TryHereOnly(wxEvent&) ()
#2  0x00000000032e2b1a in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#3  0x00000000032e2bc1 in wxEvtHandler::ProcessEvent(wxEvent&) ()
#4  0x00000000032e2423 in wxEvtHandler::ProcessPendingEvents() ()
#5  0x00000000031f9117 in wxAppConsoleBase::ProcessPendingEvents() ()
#6  0x00000000030fef4e in wxApp::DoIdle() ()
#7  0x00000000030ff033 in wxapp_idle_callback ()
#8  0x00007ffff6a21cb2 in g_idle_dispatch
    (source=0x7ffd6401eb30, callback=0x30ff020 <wxapp_idle_callback>, user_data=0x0) at ../glib/gmain.c:6124
#9  0x00007ffff6a22cbf in g_main_dispatch (context=0x49346b0)
    at ../glib/gmain.c:3444
#10 g_main_context_dispatch (context=0x49346b0) at ../glib/gmain.c:4162
#11 0x00007ffff6a78598 in g_main_context_iterate.constprop.0
    (context=0x49346b0, block=1, dispatch=1, self=<optimized out>)
    at ../glib/gmain.c:4238
#12 0x00007ffff6a2228f in g_main_loop_run (loop=0xa41e2e0)
    at ../glib/gmain.c:4438
#13 0x00007ffff7208f8d in gtk_main () at ../gtk/gtkmain.c:1329
#14 0x000000000310ede5 in wxGUIEventLoop::DoRun() ()
#15 0x000000000322d90d in wxEventLoopBase::Run() ()
--Type <RET> for more, q to quit, c to continue without paging--c
#16 0x00000000031616c9 in wxDialog::ShowModal() ()
#17 0x0000000000f69b63 in Slic3r::GUI::DPIAware<wxDialog>::ShowModal() ()
#18 0x00000000032e2572 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#19 0x00000000032e2917 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#20 0x00000000032e2a70 in wxEvtHandler::TryHereOnly(wxEvent&) ()
#21 0x00000000032e2b1a in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#22 0x00000000032e2bc1 in wxEvtHandler::ProcessEvent(wxEvent&) ()
#23 0x00000000032e2423 in wxEvtHandler::ProcessPendingEvents() ()
#24 0x00000000031f9117 in wxAppConsoleBase::ProcessPendingEvents() ()
#25 0x00000000030fef4e in wxApp::DoIdle() ()
#26 0x00000000030ff033 in wxapp_idle_callback ()
#27 0x00007ffff6a21cb2 in g_idle_dispatch (source=0xa3e91c0, callback=0x30ff020 <wxapp_idle_callback>, user_data=0x0) at ../glib/gmain.c:6124
#28 0x00007ffff6a22cbf in g_main_dispatch (context=0x49346b0) at ../glib/gmain.c:3444
#29 g_main_context_dispatch (context=0x49346b0) at ../glib/gmain.c:4162
#30 0x00007ffff6a78598 in g_main_context_iterate.constprop.0 (context=0x49346b0, block=1, dispatch=1, self=<optimized out>) at ../glib/gmain.c:4238
#31 0x00007ffff6a2228f in g_main_loop_run (loop=0x9f9cdc0) at ../glib/gmain.c:4438
#32 0x00007ffff7208f8d in gtk_main () at ../gtk/gtkmain.c:1329
#33 0x000000000310ede5 in wxGUIEventLoop::DoRun() ()
#34 0x000000000322d90d in wxEventLoopBase::Run() ()
#35 0x00000000031f9f7d in wxAppConsoleBase::OnRun() ()
#36 0x0000000003283c94 in wxEntry(int&, wchar_t**) ()
#37 0x000000000100b70b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#38 0x000000000086efa3 in Slic3r::CLI::run(int, char**) ()
#39 0x00000000007cf793 in main ()
```

jwise, 2023-02-06:
yes, agree that that is a dup of #1196 , I would believe that if you log out and log back in again you will see that happen

lanewei120, 2023-02-08:
we just submitted a fix for the crash when switch between printer(some also happens after print/send_to_print)
the issue with above callstack is very likely fixed by this patch
https://github.com/bambulab/BambuStudio/commit/9fddc49787e7af46b2af2538fe8746fc0fb6e14d

please have a try, it is already in branch v1.4.2

andypiper, 2023-02-15:
I'm failing to get this (HEAD) to build on a new install of Fedora 37 - hitting an issue with boost:

```
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindBoost.cmake:2377 (find_package_handle_standard_args)
  CMakeLists.txt:343 (find_package)
```

I'm following the steps in the [build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide).

lanewei120, 2023-02-15:
> I'm failing to get this (HEAD) to build on a new install of Fedora 37 - hitting an issue with boost:
> 
> ```
> CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
>   Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
>   log locale regex chrono atomic date_time iostreams) (Required is at least
>   version "1.66.0")
> Call Stack (most recent call first):
>   /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
>   /usr/share/cmake/Modules/FindBoost.cmake:2377 (find_package_handle_standard_args)
>   CMakeLists.txt:343 (find_package)
> ```
> 
> I'm following the steps in the [build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide).
> 
> However, I am able to start the Fedora-built AppImage posted above by @spuiuk (THANK YOU), which is pleasing. I'll test this some more tomorrow when I'm in the studio with my X1C

it seems the dependency directory not set correctly
could you show your building commands?

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120