bambulab/BambuStudio #762: Add ability to send ready to print projects to a printer which is currently in use

AimeeAllen, 2022-12-04:
When a project in Bambu Studio is ready for printing, there are currently two options:
a) send direct to a printer to print **immediately**
b) save 'locally' to an SD card to then take to the printer later

Sometimes I have several projects setup and ready to print at the same time, but not enough printers. It would be ideal to be able to send several projects wirelessly to the printer at the same time, print one, and save the rest for later. The current option to save to a local SD card doesn't work well for me as I have to keep switching the SD cards in the printer as there must be a card in the printer to record video of my current print, and so I can not use this same card (at the same time) to save my Bambu Studio files ready to print.

Perhaps if a user has an SD card in the printer, Bambu studio could get access to save wirelessly to this so the file is already on the printer waiting for the user. I would be happy with having to manually start the jobs at the printer once they were saved. Even better though, if once saved to the printer, a user could remotely select the saved jobs from Bambu Handy/Studio to start them on that printer.

Merlwynd, 2022-12-04:
I support this request...especially last sentence as it allows remote start from Bamby Handy when away from computer/printer.

AbeFM, 2022-12-07:
+1, mentioned elsewhere

Ali229, 2022-12-29:
+1 need this feature